ARMmbed/mbed-os #10007: CY8CKIT_062_WIFI_BT: IAR timeout for components-target_psa-tests-compliance_crypto-test

0xc0170, 2019-03-08:
### Description

```
mbedgt: test suite 'components-target_psa-tests-compliance_crypto-test_c035' ......................... TIMEOUT in 148.44 sec
	test case: 'Check1' .......................................................................... ERROR in 0.00 sec
```

More info filled by psa team. (to be added)

@ARMmbed/team-cypress @ARMmbed/mbed-os-psa 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

maclobdell, 2019-05-01:
@mikisch81 @orenc17 @avolinski  - do you have any more information or update on this issue?

orenc17, 2019-05-01:
running fine in 5.12.2 all toolchains

maclobdell, 2019-05-08:
@0xc0170  - has this issue been resolved?  Can it be closed?

alzix, 2019-05-12:
we have not reproduced it on our side.
tested we all toolchains on our CI - i suggest to close it.

maclobdell, 2019-05-15:
closing, please re-open if needed

closed 2019-05-15 by maclobdell