ARMmbed/mbed-os #10009: Override mbed flash behavior for Greentea

janjongboom, 2019-03-08:
### Description

Quite a lot of the Mbed OS tools require a way to flash new binaries on boards, but they all rely on using mass-storage device mode. This is fine for development boards, but when you move away from them (towards a real product) you often don't have this. E.g. many of the cheap JLINK adapters don't support MSD, or maybe you're using Atmel EDBG. Is there any way to override the flashing behavior without modifying Mbed OS itself? I've ran into this issue on three occassions in the last two weeks, with @desowin from Etteplan, with @martinichka from TWTG, and with The Things Network.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

desowin, 2019-03-08:
There is related issue in mbed-cli https://github.com/ARMmbed/mbed-cli/issues/475

cmonr, 2019-03-13:
CC @ARMmbed/mbed-os-tools

theotherjimmy, 2019-03-13:
@janjongboom My first impression is that you would have to break `mbed test` into`mbed test --compile` followed by `mbed test --run` and specify the flashing method to `mbed test --run`, which should be able to take greentea arguments directly.

loverdeg-ep, 2019-03-13:
@janjongboom Add @loverdeg-ep, @trowbridgec, @farrenv, from Embedded Planet to your list.

We've been dealing with this for months.
Please see https://github.com/ARMmbed/DAPLink/issues/577#issuecomment-460822443

janjongboom, 2019-03-14:
@theotherjimmy Or let me specify a flash python script in the custom_targets.json or something. That would help.

closed 2019-10-31 by ciarmcom