ARMmbed/mbed-os #10013: F446RE never goes into deep sleep

janjongboom, 2019-03-08:
### Description

I have a NUCLEO-F446RE development board, running with TICKLESS enabled, running basic blinky. This board never goes into deep sleep because `MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER` is declared. What is this setting and why does it block deep sleep? I thought all STM32's now support tickless mode and should go to deep sleep automatically?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-03-08:
@ARMmbed/team-st-mcd

jeromecoutant, 2019-03-08:
@c1728p9

c1728p9, 2019-03-08:
Hi @janjongboom on devices which require the low power ticker wrapper, interrupt latency when running tickless off of the low power ticker is bad enough to drop bytes when running serial at a baud of 115200. I profiled the low power ticker and low power ticker wrapper code to try and correct the high interrupt latency, but I couldn't find any further optimizations to correct this.

To workaround this #9785 makes devices which make use of the low power ticker wrapper use the microsecond ticker to run tickless by default. Applications which don't require high baudrate serial can set `MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER` to false to bring back deep sleep.

janjongboom, 2019-03-08:
@c1728p9 Interesting. I was actually running at baud rate 9,600 so no issues for me. I'll try the workaround. Additionally, @ARMmbed/mbed-docs can we make sure this is documented somewhere? Right now it is not.

janjongboom, 2019-03-12:
@c1728p9 @jeromecoutant With this macro enabled the application immediately hard faults on my F446RE. No output on serial, and I don't have a debugger to check what's going on...

iriark01, 2019-03-12:
@AnotherButler please can you see about adding this to the 5.13 plan?

closed 2019-10-31 by ciarmcom