ARMmbed/mbed-os #10024: Workaround for Ublox G450 firmware issue

NeilMacMullen, 2019-03-10:
### Description
When using a TCP socket, the Ublox G450 appears to suffer from a firmware bug where, if more bytes are requested with USORD than are available in the internal modem buffer, and then more data is received over the air, an unsolicited response can be generated.  Eg.

> 
>   ---> Client requests 20K bytes from remote server 
>   <--- +UUSORD 1500
>   ---> AT+USORD 1000
>   <--- +USORD 1000 ...
>   <---- +UUSORD 500
>   ---> AT+USORD 1000
>   <--- +USORD 500 ...
>  <---- more data received from network.
>   <--- +USORD 1000....  //'unsolicited' response
>   <--- ++UUSORD ...
> 
The Ublox AT command manual says 

> If the number of data bytes requested to be read from the buffer is bigger than the number of bytes stored in the buffer only the available amount of data bytes will be read.

and the stack has been written with this assumption but the following code appears to be a safer implementation for UBLOX_AT_CellularStack::socket_recvfrom_impl in UBLOX_AT_CellularStack.cpp 
```

 } else if (socket->proto == NSAPI_TCP) {
        while (success && (size > 0)) {

            read_blk = MIN(read_blk,UBLOX_MAX_PACKET_SIZE);
            read_blk = MIN(read_blk,size);
            read_blk = MIN(read_blk,socket->pending_bytes);
            if (socket->pending_bytes > 0) {
                _at.cmd_start("AT+USORD=");
                _at.write_int(socket->id);
                _at.write_int(read_blk);
                _at.cmd_stop();

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X Bug

0xc0170, 2019-03-11:
cc @ARMmbed/mbed-os-wan @ARMmbed/team-ublox

bqam-ublox, 2019-03-26:
@NeilMacMullen thanks for the input. I have checked with SARA-G450 FW:09.02,A02.01 and I dont see any problem if I dont wait for the +UUPSDA URC. Infact, this URC is present for U201 as well and is received as soon as the UPSDA command is sent (for both modules). I checked with m-center (AT terminal) and I also used UPING command immediately after the context activation, able to see ping responses from 8.8.8.8. Please let me know how I can replicate this issue.

Moreover, as you might already know, the cellular state machine in mbed always makes sure that modem wakes up, initializes, perform SIM checks and registers on the network. For this reason, the URC can be ignored but it makes perfect sense to parse it for result (and in that case, we will be waiting for it as well).

0xc0170, 2019-10-25:
@ARMmbed/team-ublox Please review

fahim-ublox, 2019-10-25:
Looks fine to me

0xc0170, 2019-10-25:
@NeilMacMullen Please send a pull request to fix the issue

adbridge, 2021-06-07:
> @NeilMacMullen Please send a pull request to fix the issue

As there has been no response to this I am closing this issue.

closed 2021-06-07 by adbridge