ARMmbed/mbed-os #10032: Greentea tests fail with MTS_MDOT_F411RE (sync failure)

kimlep01, 2019-03-11:
### Description

Our Greentea based Lorawan tests started to fail last week 7th of March with target: MTS_MDOT_F411RE. Reason is sync failure after flashing. After some debugging I noticed that none of the mbed-os Greentea tests work anymore with the same target. I ran git bisect with mbed-os and got the following commit:

aeeb43fb3c0f7f955a2856c7ef2c58f37bbbd2dd` is the first bad commit
commit aeeb43fb3c0f7f955a2856c7ef2c58f37bbbd2dd
Author: Jimmy Brisson <theotherjimmy@gmail.com>
Date:   Thu Feb 14 14:39:56 2019 -0600

    Simplify hook tool implementation
    
    The "hook tools" were capable of hooking into many commands run by the build system.
    To my knowlage, the only hook is the "post-build-hook". The post build hook could be
    easier to reason about if the implementation is specialized for just post-build
    hooking.
    
    This commit make it much easier to point out where post build hooks are called by
    making the call explicit.

:040000 040000 19fbd747c0de648bf2edb13c5368f3f444209875 3e8d359c14997a4044b23db3ef0ea7a8768daabd M	tools

Steps to reproduce:
- Clone mbed-os
- git checkout aeeb43fb3c0f7f955a2856c7ef2c58f37bbbd2dd (or newer)
- mbed test --compile -n tests-mbed_drivers-echo -m MTS_MDOT_F411RE -t GCC_ARM -c
- mbedgt -n tests-mbed_drivers-echo -grm MTS_MDOT_F411RE:raas_client:ruka.mbedcloudtesting.com:80 -V
- checkout any earlier commit and repeat

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

kimlep01, 2019-03-11:
Update from our CI: MTB_MTS_DRAGONFLY has the same exact issue.

0xc0170, 2019-03-11:
@ARMmbed/team-multitech Please review

klaas019, 2019-03-11:
Reviewing...
In my review I want to understand how the commit that broke it was merged. Aren't the Greentea tests run on every pr?

klaas019, 2019-03-11:
The commit broke running blinky on the mDot.

klaas019, 2019-03-11:
The bootloader for the mDot is no longer being attached post build. @0xc0170

klaas019, 2019-03-12:
Line 405 in mbed-os/tools/targets/__init__.py is causing the issue.
`toolchain_labels = set(c.__name__ for c in getmro(toolchain.__class__))
`
toolchain_labels is empty now after removing hooks.py

klaas019, 2019-03-12:
Possible fix that works for the mDot  `toolchain_labels = {toolchain}`
I have not tested this fix for other boards, configurations. @maclobdell

0xc0170, 2019-03-12:
> The commit broke running blinky on the mDot.

Which commit? I dont see any reference

reissjason, 2019-03-12:
The first commit [https://github.com/ARMmbed/mbed-os/commit/aeeb43fb3c0f7f955a2856c7ef2c58f37bbbd2dd](aeeb43f) broke.
The toolchain variable is now a string instead of a class. i.e. "GCC_ARM"
The set created by `toolchain_labels = set(c.__name__ for c in getmro(toolchain.__class__))` ended up being `{"basicstring", "object", "str"}`.

0xc0170, 2019-03-12:
@theotherjimmy Can you please review ^^ ?

theotherjimmy, 2019-03-12:
guys I don't think that commit stands alone. It was part of a PR which may have been a red-green refactor cycle.

theotherjimmy, 2019-03-12:
Also, github is doing weird redirects on some of the links.

theotherjimmy, 2019-03-12:
@reissjason Which `toolchain` are you referring to? That variable name is in basically every file.

klaas019, 2019-03-12:
@theotherjimmy Line 405 in `mbed-os/tools/targets/__init__.py` is
`toolchain_labels = set(c.__name__ for c in getmro(toolchain.__class__))`

theotherjimmy, 2019-03-12:
```diff
diff --git a/tools/targets/__init__.py b/tools/targets/__init__.py
index b122ef1adb..589ba5b956 100644
--- a/tools/targets/__init__.py
+++ b/tools/targets/__init__.py
@@ -356,7 +356,7 @@ class Target(namedtuple("Target", "name json_data resolution_order resolution_or
     def is_PSA_non_secure_target(self):
         return 'NSPE_Target' in self.labels

-    def get_post_build_hook(self, toolchain):
+    def get_post_build_hook(self, toolchain_list):
         """Initialize the post-build hooks for a toolchain. For now, this
         function only allows "post binary" hooks (hooks that are executed
         after the binary image is extracted from the executable file)
diff --git a/tools/toolchains/mbed_toolchain.py b/tools/toolchains/mbed_toolchain.py
index ba67ec3060..8016dc689b 100755
--- a/tools/toolchains/mbed_toolchain.py
+++ b/tools/toolchains/mbed_toolchain.py
@@ -136,8 +136,6 @@ class mbedToolchain:
         self.target = target
         self.name = self.__class__.__name__

-        # compile/assemble/link/binary hooks
-        self._post_build_hook = target.get_post_build_hook(self.name)

         # Toolchain flags
         self.flags = deepcopy(build_profile or self.profile_template)
@@ -754,9 +752,13 @@ class mbedToolchain:
         else:
             updatable = None

-        if self._post_build_hook:
+        # compile/assemble/link/binary hooks
+        post_build_hook = self.target.get_post_build_hook(
+            self._get_toolchain_labels()
+        )
+        if post_build_hook:
             self.progress("post-build", name)
-            self._post_build_hook(self, r, elf, full_path)
+            post_build_hook(self, r, elf, full_path)
         # Initialize memap and process map file. This doesn't generate output.
         self.mem_stats(mapfile)

         ```

Does this fix it?

theotherjimmy, 2019-03-12:
... github is not happy today.

theotherjimmy, 2019-03-12:
```diff
diff --git a/tools/targets/__init__.py b/tools/targets/__init__.py
index b122ef1adb..512c00884f 100644
--- a/tools/targets/__init__.py
+++ b/tools/targets/__init__.py
@@ -356,7 +356,7 @@ class Target(namedtuple("Target", "name json_data resolution_order resolution_or
     def is_PSA_non_secure_target(self):
         return 'NSPE_Target' in self.labels

-    def get_post_build_hook(self, toolchain):
+    def get_post_build_hook(self, toolchain_labels):
         """Initialize the post-build hooks for a toolchain. For now, this
         function only allows "post binary" hooks (hooks that are executed
         after the binary image is extracted from the executable file)
@@ -401,7 +401,6 @@ class Target(namedtuple("Target", "name json_data resolution_order resolution_or
                 ("not found in class '%s'" %  class_name))
         # Check if the hook specification also has toolchain restrictions
         toolchain_restrictions = set(hook_data.get("toolchains", []))
-        toolchain_labels = set(c.__name__ for c in getmro(toolchain.__class__))
         if toolchain_restrictions and \
            not toolchain_labels.intersection(toolchain_restrictions):
             return
diff --git a/tools/toolchains/mbed_toolchain.py b/tools/toolchains/mbed_toolchain.py
index ba67ec3060..8016dc689b 100755
--- a/tools/toolchains/mbed_toolchain.py
+++ b/tools/toolchains/mbed_toolchain.py
@@ -136,8 +136,6 @@ class mbedToolchain:
         self.target = target
         self.name = self.__class__.__name__

-        # compile/assemble/link/binary hooks
-        self._post_build_hook = target.get_post_build_hook(self.name)

         # Toolchain flags
         self.flags = deepcopy(build_profile or self.profile_template)
@@ -754,9 +752,13 @@ class mbedToolchain:
         else:
             updatable = None

-        if self._post_build_hook:
+        # compile/assemble/link/binary hooks
+        post_build_hook = self.target.get_post_build_hook(
+            self._get_toolchain_labels()
+        )
+        if post_build_hook:
             self.progress("post-build", name)
-            self._post_build_hook(self, r, elf, full_path)
+            post_build_hook(self, r, elf, full_path)
         # Initialize memap and process map file. This doesn't generate output.
         self.mem_stats(mapfile)
```

theotherjimmy, 2019-03-12:
I'm pretty sure that's the solution. PR inbound.

PR #10065 (2019-03-17): Correct post-build-hook detection logic
### Description

Fixes #10032

The problem was that post build hook detection logic expected two things:
 * The toolchain argument was a toolchain object
 * That it could recreate toolchain labels itself

Neither of these assupmtions are correct.

I changed the logic to take in a list of toolchain labels, and delegated
the creating of toolchain labels to the toolchain. Then I cleaned up all
the pylint warnings.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
  [... cut]
closed 2019-03-17 by cmonr