ARMmbed/mbed-os #10035: mbed-os/features/frameworks/TARGET_PSA/ compiled with error

TaniaMirzin, 2019-03-11:
### https://github.com/ARMmbed/mbed-os/#525d46378e8ebe5a9f71e79ced203adb1f162a65
 cimpiles TARGET_PSA framework and causes compilation errors

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jenia81, 2019-03-11:
@0xc0170 Pelion Client isn't compiled with Mbed OS master. This is blocker for us

0xc0170, 2019-03-11:
@ARMmbed/mbed-os-psa please review

alzix, 2019-03-11:
Seems like crypto misconfiguration.
@Patater , please advise.

Patater, 2019-03-11:
The issue appears to be arising from `pal_crypto_config_check.h`, but that's not an Mbed Crypto nor Mbed TLS file. For example, it's complaining about `ARCH_TEST_HMAC` missing dependencies, but `ARCH_TEST_HMAC` is not present in Mbed Crypto nor Mbed TLS.

Is anyone from the PAL/Client team available to advise? I'm happy to help if I can be of any assistance.

adbridge, 2019-03-11:
@JanneKiiskila @jenia81 Looks like PAL/client issue from Jaeden's comment above ?

orenc17, 2019-03-11:
a fix is on the way from @netanelgonen

netanelgonen, 2019-03-11:
@TaniaMirzin can you please a look?

TaniaMirzin, 2019-03-12:
@netanelgonen It's ok now. Thanks

closed 2019-03-12 by TaniaMirzin