ARMmbed/mbed-os #10040: Had a compiler (online) fault with uint32_t invalid pointer type nvic_setvector...looking around net seems this was changed a year or so ago but the onine compiler doesnt have the cast ...?

psionprime, 2019-03-11:
Had acompiler (online) fault with uint32_t invalid pointer type nvic_setvector...looking around net seems this was changed a year or so ago but the onine compiler doesnt have the cast ...?

_Originally posted by @psionprime in https://github.com/ARMmbed/mbed-os/pull/4308#issuecomment-471609558_

Target Max32630Fthr

linlingao, 2019-03-11:
@psionprime How did you come to the conclusion the online compiler doesn't have the cast?
Can you post the actual build failure you're seeing? Did you try to import a program or you're building your own code?

psionprime, 2019-03-12:
Target MAX32630FTHR
https://os.mbed.com/platforms/MAX32630FTHR/

Import example program:
https://os.mbed.com/compiler/#import:/users/switches/code/FTHR_OLED/
(https://os.mbed.com/users/switches/code/FTHR_OLED/)
Check "update all libraries to latest version".

Click compile.
>Error: Argument of type "std::uint32_t" is incompatible with parameter of type "irq_fn" in "USBDevice/USBDevice/USBHAL_Maxim.cpp", Line: 144, Col: 31
(line 144) NVIC_SetVector(USB_IRQn, (uint32_t)&_usbisr);

I just bought this so I don't have a history of knowledge to draw on. Googling brought  up only a couple of comments. One was the uint32_t cast was added, on purpose a year or so ago, pulled and adopted into the main release (ref this repository). Another comment was along the lines, of hey it was working fine now it is broke...
(ref: https://os.mbed.com/questions/61755/USBDevice-USBMSD_SD/)

After exporting the example project to be able to better search the code, I see the cast of unint32_t is not there. The mbed os repository has it.

I'm notifying those with a better understanding of the interaction of the online compiler, mbed os git repository, example project imports that there is some kind of inconsistency here that I don't know how to fix (the integrated systems level) At the code level I can start adding or subtracting casts ;) but that doesn't fix the integration nor my not understand a best practice of whether the cast is to be there at all. I'm guessing yes since it was reviewed and adopted but I'm new to mbed and this target.

Thanks for responding !

linlingao, 2019-03-12:
@psionprime There's a known issue on the online compiler. The "update all libraries to latest version" and "Update all" do not always work. We're seeing it on Mbed 2. This issue has been reported to the our web team.
There's an easy workaround. If you right click on mbed-os, then click "update", your mbed-os will be updated to the latest version. Your project will build, I've verified it.
Please let me know if this workaround works for you.

psionprime, 2019-03-12:
Built with only warnings. Thanks !

closed 2019-03-12 by psionprime