ARMmbed/mbed-os #10051: NUCLEO-F446RE hard faults when having sleep tracing enabled

janjongboom, 2019-03-12:
### Description

Verified on Mbed OS 5.11.5. On the NUCLEO-F446RE the board hard faults almost immediately when enabling sleep tracing:

```
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x8002239
Error Value: 0x1
Current Thread: rtx_idle  Id: 0x200011E4 Entry: 0x800220D StackSize: 0x300 StackMem: 0x20001528 SP: 0x2001FF3C 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020125
-- MbedOS Error Info --
```

My application:

```cpp
#include "mbed.h"
#include "mbed_stats.h"

static DigitalOut led(LED1);

void print_stats()
{
    mbed_stats_cpu_t stats;
    mbed_stats_cpu_get(&stats);
}

int main() {
    while (1) {
        mbed_stats_cpu_t stats;
        mbed_stats_cpu_get(&stats);
        printf("Uptime: %llu ", stats.uptime);
        printf("Idle: %llu ", stats.idle_time);
        printf("Sleep time: %llu ", stats.sleep_time);
        printf("Deep Sleep: %llu\n", stats.deep_sleep_time);

        led = !led;
	    printf("LED is %d\n", led.read());

        // print_stats();
        wait_ms(2000);
    }
}
```

My mbed_app.json:

```json
{
    "macros": [
        "MBED_TICKLESS=1",
        "MBED_CPU_STATS_ENABLED=1",
        "MBED_SLEEP_TRACING_ENABLED=1"
    ]
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

screamerbg, 2019-03-14:
@linlingao There are several topics related to TICKLESS and sleep issues, some of which are related to Mbed OS - https://github.com/ARMmbed/mbed-os/issues/10085, https://github.com/ARMmbed/mbed-os/issues/10079, https://github.com/ARMmbed/mbed-os/issues/10080. Could we remove the "devices: st" label until there is clarity about the issues inside Mbed OS with tickless and sleep, as this may end up being Mbed OS problem, not ST problem?

jeromecoutant, 2019-04-08:
> until there is clarity about the issues inside Mbed OS with tickless and sleep, as this may end up being Mbed OS problem

It seems that this issue is not assigned to anybody yet ?

linlingao, 2019-04-08:
@jeromecoutant The issue has been assigned to the HAL team.

LMESTM, 2019-06-21:
#10702 and #10781 may solve this issue.

jeromecoutant, 2019-06-24:
I think we could close this one ?
Thx

closed 2019-06-24 by 0xc0170