ARMmbed/mbed-os #10052: mbed-os-example-blinky not compiling against mbed-os-5.12

JanneKiiskila, 2019-03-12:
### Description

mbed-os-5.12 branch at commit 7dd791e2f992d872b2dab4a7f061d05f7283090a fails to compile.

mbed version 1.9.1
pip 19.0.3
gcc arm GCC_ARM_PATH=/home/jankii01/gcc-arm-none-eabi-6-2017-q2-update/bin

```
mbed-os-example-blinky (current master),
mbed deploy
cd mbed-os
git checkout mbed-os-5.12
cd ..
mbed compile -m K64F -t GCC_ARM
```

fails to compile.

```
jankii01@ubuntu:~/mbed/mbed-os-example-blinky$ mbed compile -m K64F -t GCC_ARM
[mbed] Working path "/home/jankii01/mbed/mbed-os-example-blinky" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: pycryptodome, cmsis_pack_manager
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
Traceback (most recent call last):
  File "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os/tools/make.py", line 41, in <module>
    from tools.options import get_default_options_parser
  File "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os/tools/options.py", line 24, in <module>
    from .toolchains import TOOLCHAINS
  File "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os/tools/toolchains/__init__.py", line 19, in <module>
    from . import mbed_toolchain, arm, gcc, iar
  File "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os/tools/toolchains/mbed_toolchain.py", line 48, in <module>
    from ..config import (ConfigException, RAM_ALL_MEMORIES, ROM_ALL_MEMORIES)
  File "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os/tools/config/__init__.py", line 37, in <module>
    from ..arm_pack_manager import Cache
  File "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os/tools/arm_pack_manager/__init__.py", line 22, in <module>
    from cmsis_pack_manager import Cache as _Cache
ImportError: No module named cmsis_pack_manager
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/jankii01/mbed/mbed-os-example-blinky"
       Command: "/usr/bin/python -u /home/jankii01/mbed/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

### Issue request type
-->
    [ ] Question
    [ ] Enhancement
    [x] Bug

JanneKiiskila, 2019-03-12:
Full pip list.

```
jankii01@ubuntu:~/mbed/mbed-os-example-blinky$ pip list
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Package                       Version    
----------------------------- -----------
adium-theme-ubuntu            0.3.4      
alabaster                     0.7.10     
appdirs                       1.4.3      
argcomplete                   1.9.3      
asn1ate                       0.6.0      
asn1crypto                    0.24.0     
Babel                         2.5.3      
backports-abc                 0.5        
beautifulsoup4                4.6.0      
certifi                       2018.1.18  
cffi                          1.11.4     
chardet                       3.0.4      
click                         6.7        
colorama                      0.3.9      
coloredlogs                   9.0        
cryptography                  2.2.2      
docutils                      0.14       
ecdsa                         0.13       
enum34                        1.1.6      
fasteners                     0.14.1     
flex                          6.12.0     
funcsigs                      1.0.2      
functools32                   3.2.3.post2
future                        0.16.0     
futures                       3.2.0      
fuzzywuzzy                    0.16.0     
hgapi                         1.7.4      
humanfriendly                 4.8        
icetea                        1.0.2      
idna                          2.6        
imagesize                     0.7.1      
intelhex                      2.2.1      
intervaltree                  2.1.0      
ipaddress                     1.0.19     
Jinja2                        2.10       
jsonmerge                     1.4.0      
jsonpointer                   1.14       
jsonschema                    2.6.0      
junit-xml                     1.8        
livereload                    2.5.1      
lockfile                      0.12.2     
Logbook                       1.1.0      
lxml                          4.1.1      
manifest-tool                 1.4.6      
Markdown                      2.6.11     
MarkupSafe                    1.0        
mbed-cli                      1.9.1      
mbed-cloud-sdk                2.0.1      
mbed-flasher                  0.9.2      
mbed-greentea                 1.3.3      
mbed-host-tests               1.3.0      
mbed-ls                       1.6.2      
mbed-test-wrapper             1.0.0      
meld                          3.14.2     
mercurial                     3.7.3      
mkdocs                        0.17.2     
mock                          2.0.0      
monotonic                     1.4        
pathlib                       1.0.1      
pbr                           3.1.1      
pip                           19.0.3     
prettytable                   0.7.2      
project-generator             0.8.16     
project-generator-definitions 0.2.35     
protobuf                      3.5.2.post1
psutil                        5.4.3      
py                            1.5.2      
pyasn1                        0.2.3      
pycparser                     2.18       
pyelftools                    0.25       
PyGithub                      1.35       
Pygments                      2.2.0      
pygobject                     3.20.0     
PyJWT                         1.5.3      
pyocd                         0.14.3     
pyOpenSSL                     17.5.0     
pyparsing                     2.2.0      
pyserial                      3.4        
pyshark                       0.3.7.11   
pyshark-legacy                0.3.8      
python-dateutil               2.8.0      
python-dotenv                 0.10.1     
pytz                          2017.3     
pyusb                         1.0.2      
PyYAML                        4.2b1      
requests                      2.20.1     
rfc3987                       1.3.7      
semantic-version              2.6.0      
semver                        2.7.9      
setuptools                    38.4.0     
singledispatch                3.4.0.3    
six                           1.11.0     
snowballstemmer               1.2.1      
sortedcontainers              2.1.0      
Sphinx                        1.6.6      
sphinxcontrib-websupport      1.0.1      
strict-rfc3339                0.7        
tornado                       4.5.3      
trollius                      1.0.4      
typing                        3.6.4      
unity-lens-photos             1.0        
urllib3                       1.23       
validate-email                1.3        
valinor                       0.0.15     
websocket-client              0.46.0     
wheel                         0.30.0     
xmltodict                     0.11.0     
yattag                        1.9.2 
```

teetak01, 2019-03-12:
@JanneKiiskila I think you need to run:
```
pip install -r mbed-os/requirements.txt
```
There is some new mandatory pip-components introduced in 5.12. The auto-install probably fails because you might have globally installed pip which would require sudo-access.

JanneKiiskila, 2019-03-12:
That indeed does resolve the issue. This is likely a documentation task then.

0xc0170, 2019-03-12:
cc @ARMmbed/mbed-os-tools

linlingao, 2019-03-12:
I see warnings printed:
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: pycryptodome, cmsis_pack_manager
       You can install all missing modules by running "pip install -r requirements.txt" in "/home/jankii01/mbed/mbed-os-example-blinky/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
@JanneKiiskila Do you still feel like further action is needed?

JanneKiiskila, 2019-03-13:
We should note in the release notes, that with Mbed OS 5.12 release the requirements for Python modules have changed, please do this and that. (And we should really advice people how to use `virtualenv` to get rid of the module dependency issues9:

jupe, 2019-03-25:
same happens for me but `pip install -r mbed-os/requirements.txt`. After upgrading mbed-cli building started to work.

closed 2019-04-29 by ciarmcom