ARMmbed/mbed-os #10054: Homekit ADK accessory with iPhone 8 and iPhone X with 5.12.0 mbed

akkishore1, 2019-03-12:
Hi,
HomeKit ADK disconnects immediately when we try with iPhone8 or iPhone X with iOS version 12.1.4 (16D57) when using 5.12.0 of mbed
However Homekit ADK works fine with same iPhone8 and iPhoneX with iOS version 12.1.4 (16D57) with mbedOS version 5.11.4.

There seems to be lot of changes in the Cordio stack between 5.11.4  and 5.12.0. Is there any additional API needs to be taken care in latest Cordio Stack.

Additional Info:
Homekit ADK works fine with iPhone 7 with iOS version 10.3.1 (14E304) and iPhone5s with iOS version 12.1.4 with mbed version 5.12.0

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

desmond-blue, 2019-03-13:
Would it be okay to share the sniffer trace of disconnecting scenario? Thanks.

akhilcypress, 2019-03-13:
[sniffer.zip](https://github.com/ARMmbed/mbed-os/files/2960463/sniffer.zip)
failure case sniffer log

akhilcypress, 2019-03-13:
We are also facing similar issue with GattServer example app,
Logs from GattServer with iphone X + iOS-12.1.4 

Starting GattServer
Ble process started.
Ble instance initialized
Advertising started.
Adding demo service
clock service registered
service handle: 11
        hour characteristic value handle 13
        minute characteristic value handle 16
        second characteristic value handle 19
Connected. 
-- end of log 

Usecase: BLE_GattServer + iPhoneX (iOS 12.1.4) with LightBlue app
-        iphoneX able to discover the GattServer device via LightBlue app
-	 tap to connect GattServer device
-	On iPhone side it is showing “connecting” -> “Interrogating”  then pop up message “Connection Alert” Timeout.
- No disconnection or any other logs from serial console, It appears the stack just hanged after this

desmond-blue, 2019-03-13:
From the trace, the communication is gone after phy_update_ind, so I think all examples have the same issue, and that’s why iPhone 7 doesn’t has the issue because it’s bt4.2.

desmond-blue, 2019-03-14:
Hi @akhilcypress I just raise a PR #10092 for this, could you check if this solve your issue? Thanks.

akkishore1, 2019-03-14:
With this fix the issue is solved

PR #10092 (2019-03-17): BLE:fix on_phy_update_complete function
### Description

Fix #10054.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
closed 2019-03-17 by cmonr