ARMmbed/mbed-os #10057: STM32 DMA support

embeddedteam103, 2019-03-12:
### Description

Hello,
Is there any plan to support DMA in STM32 devices?

We are planning to implement a fork of mbed with some DMA support for our internal usage, and would like to know what we'll have to do for a PR that will be accepted.

We are considering the following types of changes:
1. [breaking change] Extending the mbed HAL API.
2. [breaking change] Adding a new mapping in the `PeripheralNames.h` file for every target supporting DMA (mapping from peripheral to DMA stream, or something similar).
3. [non breaking change] Adding a new (C++) driver for the DMA that configures which peripheral uses the stream at a given time.
4. [non breaking change] Refactor the implementation of mbed HAL layer for STM32F4 and STM32L4 for the SPI and UART peripherals - adding DMA support.

2 and 3 are optional, but we'll definetly do 1 and 4.
We have the following questions:

1. Is there any plan to support DMA in ST already?

2. Is there any plan for a DMA driver that controls peripherals usage?

3. How likely are you to accept a PR with change 1? Will implementation for all platforms be required, or is this done by each target maintainers?

4. How likely are you to accept a PR with change 2? Will implementation for all targets be required, or is this done by each target maintainers?

5. How likely are you to accept a PR with change 3? This is a pure addition, so I assume this is less of a burden than the other types of change.

6. How likely are you to accept a PR with change 4? Other then testing, is there something else?

Note this was asked before: https://github.com/ARMmbed/mbed-os/issues/3647 https://github.com/ARMmbed/mbed-os/issues/4360 
### Issue request type

    [x ] Question
    [ ] Enhancement
    [ ] Bug

loverdeg-ep, 2019-03-12:
https://github.com/ARMmbed/mbed-os/blob/master/docs/design-documents/README.md

^ The above link would be the process to follow if they don't currently have plans.

LMESTM, 2019-03-13:
@embeddedteam103 
Speaking for @ARMmbed/team-st-mcd team: we would definitely appreciate and would accept PRs that add extra features to ST targets as long as this fits into mbed-os architecture drivers and doesn't break existing features.  We're even encourage community contributions to TARGET_STM folder.

One of the key point here is to let application select whether they want to use DMA or not, and find a way to allocate DMA channels/streams between drivers. If the API doesn't allow to select DMA, at least there should be some build-time configuration.  We will definitely review PRs and would also be able to launch automated non-regression test sessions  on our ST targets.  

Of course, this will also need to be reviewed and accepted by mbed-os maintainers.

jeromecoutant, 2019-03-27:
@TacoGrandeTX 

I think setting devices ST is not fully correct.
As you can see in the description, point 1 is "Extending the mbed HAL API."

So this issue has to be first analysed mbed-os side.

Thx

@screamerbg

TacoGrandeTX, 2019-03-27:
@jeromecoutant Yes - good point.  I've removed the label.  Thanks for pointing that out.

TacoGrandeTX, 2019-03-27:
@ARMmbed/mbed-os-hal Could someone please review and comment ?

jeromecoutant, 2019-04-23:
Hi
No explanation before closing the issue.....?
:-(

linlingao, 2019-04-23:
@jeromecoutant Sorry, the issue was closed by our bot instructed by me by mistake. I'm reopening it.

0xc0170, 2019-10-25:
The answer provided above (https://github.com/ARMmbed/mbed-os/issues/10057#issuecomment-472330362) is good.

Adding new driver feature should have support most of the targets and test coverage. Any breaking change needs careful consideration, rather do backward compatible addition.

I'll close this as answered. If you have further questions, please follow up on forum.mbed.com, thanks.

Btw, have you made any progress with any items above?

jeromecoutant, 2019-10-31:
Hi

Maybe you could have a look on a development branch from Alexandre:
https://github.com/ABOSTM/mbed-os/tree/I2C_SPI_DMA_IMPLEMENTATION_FOR_STM32L4

Implementation:
- DMA for SPI master only
- Dynamic SPI DMA/IT management : Some SPI instances can use DMA while other instances continue using Interrupts
- Code for DMA is enabled with USE_SPI_DMA macro
- DMA for I2C master only
- Dynamic I2C DMA/IT management : Some I2C instances can use DMA while other instances continue using Interrupts
- Code for DMA is enabled with USE_I2C_DMA macro

Restrictions:
- Only implemented for STM32L4R5xI.  (for example no DMAMUX on L0)
- Only SPI1 and SPI2
- Only I2C1 and I2C2
- Interrupts are still in use for SPI and I2C slave

@ARMmbed/team-st-mcd 
@bulislaw 
@MarceloSalazar

bulislaw, 2019-11-06:
Hi @embeddedteam103 proper DMA support was on my mind for quite a while now. Unfortunately Mbed team had a lot of other priorities and couldn't focus on that. The problem is rather hard in context of different vendors and HW designs and Mbed commitment to provide simple, generic APIs. Currently some of our drivers support async in rather naive way, that can be implemented either using interrupts or DMA. I acknowledge it's a simplistic approach and not a full solution, will need to be fixed sooner rather than later.
As for your question, yes I agree that design document is a good starting point. That being said we won't be able to accept a change that works for one vendor (or even subset of their targets) and ignores the rest. Even worse if it breaks other targets. What we would be more open to accept is an optional API/subsystem that can be implemented by all the vendors later.
Some time ago we announced a new governance model for Mbed, where ARM and our silicon partners collaborate in a working group, for big architectural changes, new api etc. Before accepting any such change I would need to bring the proposal there so all the partners have a say and opportunity to introduce changes.
I'm assuming you'll go with the implementation one way or another, I'd be interested to see it in any case. If you can spare time some time to prepare the design document and investigate other hardware besides ST it would be awesome.
As for our plans around DMA, we won't be able to focus on that before Mbed 6 is out maybe around 6 months from now. After that I hope we can start working on it internally.
Thanks for the interest in Mbed and I'm sorry for the issues with this ticket being repeatedly closed by our bot. (The war with the machines has started!)

closed 2021-10-20 by ciarmcom