ARMmbed/mbed-os #10058: Need a way to enable sleep tracing, but without printing stats every time

janjongboom, 2019-03-12:
### Description

I really want sleep tracing on as it tells me which locks are active / non-active. However, this also prints the active sleep locks every time the device goes into sleep, which clutters my serial output significantly and I suspect the crash in https://github.com/ARMmbed/mbed-os/issues/10051 is also because of this. Can we add an extra setting that does log sleep locks that are activated / dis-activated, but does not log the stats every time we go to sleep?

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

janjongboom, 2019-04-24:
@cmonr @bulislaw Why is this closed?

gpsimenos, 2019-06-04:
Hi Jan, thanks for engaging with Mbed OS :smiley:

I've had a look at this and I definitely agree that sleep tracing can be too verbose by default in some cases. It would be great to have an additional macro to control the printing of stats every time the device goes to sleep, but introducing a new macro to Mbed OS incurs considerable maintainance and documentation effort. Unfortunately, in this case and at this time, we have decided not to make any changes.

Although I'm sure you will have implemented your own workaround by now, below is a solution you and other people who come across this issue can implement locally.

In `plaftorm/mbed_sleep_manager.c`, at the top of the `sleep_manager_sleep_auto` function, wrap the call to `sleep_tracker_print_stats()` in an additional compiler condition, like this:

```c
#ifdef MBED_SLEEP_TRACING_ENABLED
#ifndef MBED_SLEEP_TRACING_LIGHT
    sleep_tracker_print_stats();
#endif
#endif
```

Now you will be able to simply define the `MBED_SLEEP_TRACING_LIGHT` macro alongside `MBED_SLEEP_TRACING_ENABLED` to limit the amount of traces.

Either add the macro to the `mbed_app.json` in your application,

```json
{
    "macros": [
         "MBED_SLEEP_TRACING_ENABLED",
         "MBED_SLEEP_TRACING_LIGHT"
    ],
    "target_overrides": {
        
    }
}
```

or append `-DMBED_SLEEP_TRACING_ENABLED -DMBED_SLEEP_TRACING_LIGHT` to `mbed compile`.

If this solves your issue, let me know so we can close this.

closed 2019-10-31 by ciarmcom