ARMmbed/mbed-os #10060: kv_config.cpp file uses macros for QSPI pins without MBED_CONF prefix

VairaRamasamy, 2019-03-12:
### Description

The macros used for the QSPI pins are QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, and QSPI_FLASH1_IO3. These should be instead MBED_CONF_QSPIF_QSPI_IO0, MBED_CONF_QSPIF_QSPI_IO1, MBED_CONF_QSPIF_QSPI_IO2, and MBED_CONF_QSPIF_QSPI_IO3 so that someone can override the pin setting in mbed_app.json without having to modify the header file under the target folder. 

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

dannybenor, 2019-03-28:
@VairaRamasamy Can you please explain the use case where you want to change the pin settings for a certain board? I think certain board always use the same pins for QSPI. Are the pins configurable in the hardware?

VairaRamasamy, 2019-03-28:
@dannybenor You're right that the pins are fixed for a certain board. But Cypress' PSoC 6 for example supports two sets of QSPI pins. We can choose either set by configuring it in the hardware registers. e.g. Value of 0 will choose P11_6, P11_5, P11_4, and P11_3 and value of 1 will choose P12_0, P12_1, P12_2, P12_3. 

I can think of overriding the pin settings through mbed_app.json in the case where I've a memory chip connected to the second set of pins (at least during evaluation or prototype design). Though I can still modify it in the TARGET folder, defining it through mbed_app.json makes it easy to distribute the code.

dannybenor, 2019-03-28:
Can you have a look at https://github.com/ARMmbed/mbed-os/blob/master/features/storage/system_storage/SystemStorage.cpp#L111
You can put in your mbed_app.json
"qspif.QSPI_IO0": "P12_0"
under "target_overrides" for your target

VairaRamasamy, 2019-03-29:
This is not required to be fixed to work with a QSPI flash. I can instantiate QSPIFBlockDevice with the second set of pins instead of using the get_default_instance() if needed. 

I got a compile error on kv_config.cpp when I was first working with QSPIFBlockDevice without defining the pins in PinNames.h file under TARGET folder. I only thought that it's right to use the macros with 'MBED_CONF' prefix in kv_config.cpp as this prefix is used in instantiating all other block devices in the same file. May be I'm missing something. I do not know exactly when this file will be required in a user application. I'm fine with not fixing this if you don't see it as a big problem.

dannybenor, 2019-03-31:
@VairaRamasamy The file kv_config.cpp do use the macros starting with MBED_CONF. All the parameters defined in emb_app.json will get into the compilation as a #define in the form MBED_CONF_XXXXXX_YYYYY where XXXXXX is the name of the section and YYYYY is the name of the parameter. Please have a look at https://os.mbed.com/docs/mbed-os/v5.12/reference/configuration.html

VairaRamasamy, 2019-04-01:
@dannybenor 
"The file kv_config.cpp do use the macros starting with MBED_CONF"
>> Then why are we using QSPI_FLASH1_IOx macros for QSPIF, not MBED_CONF_QSPIF_QSPI_IOx?

0xc0170, 2019-08-14:
@VairaRamasamy Can you send PR fixing this? 

Based on what I read here, the current form:

```
    static QSPIFBlockDevice bd(
        QSPI_FLASH1_IO0,
        QSPI_FLASH1_IO1,
        QSPI_FLASH1_IO2,
        QSPI_FLASH1_IO3,
        QSPI_FLASH1_SCK,
        QSPI_FLASH1_CSN,
        QSPIF_POLARITY_MODE_0,
        MBED_CONF_QSPIF_QSPI_FREQ
    );
```

could be changed to use MBED_CONF as other files. You can keep it backward compatible by checking if QSPI_FLASH1_ defined, use these pins otherwise use MBED_CONF qspi pins as requested.

cy-vaira, 2019-08-20:
@0xc0170 I'll submit the PR before end of this week.

cy-vaira, 2019-08-23:
@0xc0170 we do not need to use #ifdef since MBED_CONF_QSPIF_ are anyway pointing to QSPI_FLASH1_ as defined in mbed_lib.json at \components\storage\blockdevice\COMPONENT_QSPIF.

0xc0170, 2019-08-28:
Fixed on master

closed 2019-08-28 by 0xc0170