ARMmbed/mbed-os #10061: esp8266-driver can't compile is DEVICE_INTERRUPTIN is not enabled

jeromecoutant, 2019-03-12:
### Description

Issue is following #9857 

If you remove INTERRUPTIN from DEVICE_HAS json configuration, 
mbed-os compilation is failing:

````
$ mbed compile -t ARM -m NUCLEO_L476RG -v

[DEBUG] Output: In file included from .\components\wifi\esp8266-driver\ESP8266\ESP8266.cpp:22:
[DEBUG] Output: ./components/wifi/esp8266-driver/ESP8266/ESP8266.h:396:11: error: no type named 'UARTSerial' in namespace 'mbed'
[DEBUG] Output:     mbed::UARTSerial _serial;
[DEBUG] Output:     ~~~~~~^
[DEBUG] Output: .\components\wifi\esp8266-driver\ESP8266\ESP8266.cpp:174:34: error: use of undeclared identifier 'SerialBase'
[DEBUG] Output:         _serial.set_flow_control(SerialBase::Disabled, _serial_rts, _serial_cts);
[DEBUG] Output:                                  ^
````

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-03-21:
Hi
Any updates ?

0xc0170, 2019-03-22:
It was looked at but not fixed so far (I reviewed internal jira reference).

PR #10244 (2019-03-29): Add the INTERRUPTIN compilation guard for ESP8266
### Description

In case INTERRUPTIN flag is not set for a particular target, ESP8266 source code should not be compiled. The reason is that it relies on submodules (such as UARTSerial), which will not compile without INTERRUPTIN flag set. Thus, the compilation fails with an error without the flag.

The solution to the issue is to guard the ESP8266 with a macro checking that INTERRUPTIN flag is set.

This fixes #10061 .

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
   [... cut]
closed 2019-03-29 by adbridge