ARMmbed/mbed-os #10062: K64F Ethernet cannot connect in tickless mode

janjongboom, 2019-03-12:
### Description

Mbed OS 5.11.4 (ecb3c8c837162c73537bd0f3592c6e2a42994045), K64F, develop profile, GCC ARM 6. Running Pelion Device Management Client.

When enabling tickless mode on K64F it can no longer connect to Ethernet.

```cpp
int main() {
    NetworkInterface *net = NetworkInterface::get_default_instance();
    nsapi_error_t net_status = NSAPI_ERROR_NO_CONNECTION;
    while ((net_status = net->connect()) != NSAPI_ERROR_OK) {
        printf("Unable to connect to network (%d). Retrying...\n", net_status);
    }
```

Throws 'Unable to connect to network (-3015).'

If you wrap the call to connect with `sleep_manager_lock_deep_sleep();` and the corresponding unlock call connection works fine.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-03-12:
cc @ARMmbed/team-nxp

linlingao, 2019-08-02:
@janjongboom I finally got a chance to look into this.
First I tested the latest mbed-cloud-client-example and the latest mbed OS code. I was able to pass all these:
(1) gcc_arm develop tickless off
(2) gcc_arm develop tickless on
(3) gcc_arm debug tickless off
(4) gcc_arm debug tickless on
(5) arm develop tickless off
(6) arm develop tickless on

I tried to revert to the 5.11.4 mbed-os release, but the latest mbed-cloud-client-example no longer builds with this release.

This issue is not reproduce-able with the latest code.

linlingao, 2019-08-02:
I'll be closing this issue. If you can still reproduce this issue with the latest release, please either reopen it or create a new issue.

closed 2019-08-03 by ciarmcom

mmahadevan108, 2019-08-03:
This should be addressed in the below PR:
https://github.com/ARMmbed/mbed-os/pull/10796