ARMmbed/mbed-os #10063: cypress non-psa regresion

TaniaMirzin, 2019-03-12:
Steps to reproduce:
1. Clone https://github.com/ARMmbed/factory-configurator-client-example-internal,  revision #aa29773c060ce9d3a12c4c1fd7ed5ae5e12b76fc
2. mbed deploy
3.  mbed compile --app-config mbed_app_psa.json -m CY8CKIT_062_WIFI_BT -t GCC_ARM -DFCE_SERIAL_INTERFACE
4. Run the hex image on CYPRESS board.

Expected : the application passes all initialization and prints "Factory flow begins".
Actual : the application stacks during pal_plat_DRBGInit function in trng_get_bytes function in psa_hrng.c.

    [ ] Question
    [ ] Enhancement
    [ x] Bug

0xc0170, 2019-03-12:
cc @ARMmbed/mbed-os-psa

orenc17, 2019-03-12:
the vanilla flavor of CY8CKIT_062_WIFI_BT is without PSA.
it is not reasonable that you would be stuck in `psa_hrng.c` as you do not have PSA.

investigate 1 minute more and i find you add PSA to the target
https://github.com/ARMmbed/factory-configurator-client-example-internal/blob/aa29773c060ce9d3a12c4c1fd7ed5ae5e12b76fc/mbed_app_psa.json#L41

0xc0170, 2019-03-13:
> the vanilla flavor of CY8CKIT_062_WIFI_BT is without PSA.

If that is the case, PSA should be removed from the configuration.

closed 2019-03-27 by ciarmcom