ARMmbed/mbed-os #10069: ARM toolchain uses wrong armasm

c1728p9, 2019-03-12:
### Description

I have both ARMC5 and ARMC6 installed. When I try to compile with the latest master for ARM (which now defaults to ARMC6) I get a build error when running `armasm`. For example running `mbed compile -t ARM -m K64F -f -v` gives me:
```
Compile: armasm --cpu=Cortex-M4.fp --cpreproc --cpreproc_opts=-c,--target=arm-arm-none-eabi,-mthumb,-Os,-Wno-armcc-pragma-push-pop,-Wno-armcc-pragma-anon-unions,-DMULADDC_CANNOT_USE_R7,-fdata-sections,-fno-exceptions,-MMD,-D_LIBCPP_EXTERN_TEMPLATE(...)=,-fshort-enums,-fshort-wchar,-DMBED_TRAP_ERRORS_ENABLED=1,-mcpu=cortex-m4,-mfpu=fpv4-sp-d16,-mfloat-abi=hard,-DMBED_ROM_START=0x0,-DMBED_ROM_SIZE=0x100000,-DMBED_RAM_START=0x20000000,-DMBED_RAM_SIZE=0x30000,-DMBED_RAM1_START=0x1fff0000,-DMBED_RAM1_SIZE=0x10000,-D__CORTEX_M4,-DCPU_MK64FN1M0VMD12,-DARM_MATH_CM4,-D__FPU_PRESENT=1,-D__MBED_CMSIS_RTOS_CM,-D__CMSIS_RTOS,-DFSL_RTOS_MBED,-DMBEDTLS_PSA_CRYPTO_C,@.\BUILD\K64F\ARM\.includes_0fe1a3590d1a55ca3459e6a12cc7f55e.txt,-include,.\BUILD\K64F\ARM\mbed_config.h -o BUILD\K64F\ARM\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.o .\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.S
[DEBUG] Return: 8
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '--target=arm-arm-none-eabi'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--thumb'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-no-armcc-pragma-push-pop'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-no-armcc-pragma-anon-unions'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fdata-sections'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fno-exceptions'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--md'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fshort-enums'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fshort-wchar'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--cpu'.
[DEBUG] Output: Fatal error: C3917U: Concatenated options cannot have arguments: -cpu cortex-m4
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--fpu'.
[DEBUG] Output: Fatal error: C3917U: Concatenated options cannot have arguments: -fpu fpv4-sp-d16
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-float-abi=hard'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-I'.
[DEBUG] Output: Fatal error: A1905U: Pre-processor step failed for '.\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.S'
[DEBUG] Output: 1 Error, 0 Warnings
Traceback (most recent call last):
  File "C:\Russ\projects\Verilog\fpga\ci-test-shield-fpga\mbed-code\mbed-os\tools\make.py", line 74, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "C:\Russ\projects\Verilog\fpga\ci-test-shield-fpga\mbed-code\mbed-os\tools\build_api.py", line 486, in build_project
    objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
  File "C:\Russ\projects\Verilog\fpga\ci-test-shield-fpga\mbed-code\mbed-os\tools\toolchains\mbed_toolchain.py", line 459, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\Russ\projects\Verilog\fpga\ci-test-shield-fpga\mbed-code\mbed-os\tools\toolchains\mbed_toolchain.py", line 532, in compile_queue
    raise ToolException(err)
ToolException: Fatal error: C3900U: Unrecognized option '--target=arm-arm-none-eabi'.
```

Running the `where` command shows that armasm for ARMC5 is found first.
```
> where armasm
C:\Keil_v5\ARM\ARMCC\bin\armasm.exe
C:\Program Files\ARMCompiler6.10.1\bin\armasm.exe
```

CC @theotherjimmy 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

bridadan, 2019-04-11:
@c1728p9 is this still occurring? I've tried this recently with both ARMC6 and ARMC5 in my path and I don't believe I had any issues.

closed 2019-04-26 by c1728p9

surajdagar, 2019-08-14:
Hey,
I have exported the blinky example from the online compiler, I am building it for the target nRF52840_DK over uVision(5.26.2.0). I am also getting the same build errors. can you please check this with uVision?

0xc0170, 2019-08-23:
@surajdagar Not certain what to check. Based on the description above, this is incorrect setting in your uvision project? Check what is set there as default toolchain. Do you have ARMC6 and also ARMC5 installed ?

It could be template is wrong but without more details we are not able to fix (can't reproduce).

surajdagar, 2019-08-23:
Please follow below steps to reproduce the same.
1) import a blinky program in the online mbed compiler
2) Change target to nRF52480_DK
3) Export that program for Keil uVision
4) Open the Keil project you just downloaded and try to build that.

you will see all this crap coming in as your build output. In project options, I tried with different ARM compiler - Compiler version 5, Compiler Version 6, V5.06 update 6(build 750), V6.10.1.

0xc0170, 2019-08-26:
I am able to reproduce only if I export to `uvision5-armc5`. Looks like the profile being copied is not valid with armc5. This is confusing.

cc @ARMmbed/mbed-os-tools

0xc0170, 2019-08-28:
That should not happen (the mix of profiles). The above issue was about something else, you can create a new issue with description - "exporting to uvision-armc5 copies armc6 profile and does not compile"