ARMmbed/mbed-os #10076: Compile failed when running Greentea test with ARM_MPS2_M0

ChangwuShan, 2019-03-13:
### Description

When I trying to run greentea test with the target ARM_MPS2_M0 and ARMCC 6.11 toolchain, it failed in the compile stage.

Steps to reproduce:

1. Clone Mbed-os code (5.12 master branch)
2. https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L5096 Change this line to
"release_versions": ["2", "5"]
3. Run:  mbed test -m ARM_MPS2_M0 -t ARMC6 -n tests-mbedmicro-mbed-div

Build failures:
  * ARM_MPS2_M0::ARMC6::TESTS-MBEDMICRO-MBED-DIV
        Building project div (ARM_MPS2_M0, ARMC6)
        Scan: ARMC6
        Scan: div
        Compile [100.0%]: main.cpp
        Link: div
        Warning: L3912W: Option 'legacyalign' is deprecated.
        Error: L6218E: Undefined symbol mbed_mpu_enable_ram_xn (referred from BUILD/tests/ARM_MPS2_M0/ARMC6/platform/mbed_mpu_mgmt.o).
        Error: L6218E: Undefined symbol mbed_mpu_enable_rom_wn (referred from BUILD/tests/ARM_MPS2_M0/ARMC6/platform/mbed_mpu_mgmt.o).
        Error: L6218E: Undefined symbol mbed_mpu_init (referred from BUILD/tests/ARM_MPS2_M0/ARMC6/rtos/TARGET_CORTEX/mbed_boot.o).
        Finished: 0 information, 1 warning and 3 error messages.

It seems that M0 doesn't provide MPU, but Mbed-os call the functions about MPU when running the Greentea test, caused the build issue.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-03-13:
I am talking to @ChangwuShan . I believe there's wrong configuration in device_has. I've checked MCU header file for ARM_MPS2_M0  and it does not have MPU present, thus MPU should not be in device_has neither and no build errors regarding MPU.

ChangwuShan, 2019-03-13:
Thanks @0xc0170 , It fixed after I delete "MPU" in device_has.

closed 2019-03-13 by ChangwuShan