ARMmbed/mbed-os #10080: Bare metal mode never sleeps

janjongboom, 2019-03-13:
### Description

Looking at the power profile of NUCLEO-F446RE running Mbed OS 5.12 (7dd791e2f992d872b2dab4a7f061d05f7283090a) it never enters any sleep state. There are no sleep locks active, and I'm using a low power ticker for blinky. The sleep stats also indicate that we never go to sleep:

```
LOCK: Ticker.cpp, ln: 61, lock count: 1
UNLOCK: Ticker.cpp, ln: 32, lock count: 0
LOCK: Ticker.cpp, ln: 61, lock count: 1
UNLOCK: Ticker.cpp, ln: 32, lock count: 0
Hello world no RTOS!
Press BUTTON1 to go into standby mode
Uptime: 61 Sleep time: 0 Deep Sleep: 0
Uptime: 2076 Sleep time: 0 Deep Sleep: 0
Uptime: 4089 Sleep time: 0 Deep Sleep: 0
```

Power consumption hovers around 44 mA. Why don't we sleep in bare metal mode?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

kjbracey-arm, 2019-03-25:
I'm working on it in this PR: https://github.com/ARMmbed/mbed-os/pull/10104

> Why don't we sleep in bare metal mode?

Because non-RTOS `wait` is literally a wait. It's spinning looking at the timer. You need a sleep call, rather than a wait call. #10104 gives you `ThisThread::sleep_for` for bare metal, and deprecates `wait` - you shouldn't be "waiting" for long periods, you should be sleeping. (And having `wait` actually sleep for RTOS was inconsistent) `wait_us` and `wait_ns` remain available.

In 5.12, you can achieve sleep by using the event loop - that has non-RTOS code which is capable of sleeping, unlike `wait`. But it's not as sophisticated as the sleep code inside the RTOS idle - it doesn't have low power wakeup latency compensation. #10104 unifies and makes it work the same.

MarceloSalazar, 2020-07-28:
Closing as fixed in PR. Please re-open if continues to be a problem.

closed 2020-07-28 by MarceloSalazar