ARMmbed/mbed-os #10083: ESP8266 firmware too old, but I follow the exact instructions

janjongboom, 2019-03-13:
### Description

Using ESP8266 driver in Mbed OS 5.11 (c966348d3f9ca80843be7cdc9b748f06ea73ced0) I get the message: 

ERROR: Firmware v-1 incompatible with this driver.Update at least to v2 - https://developer.mbed.org/teams/ESP8266/wiki/Firmware-Update

However, I have followed the instructions in that page and it updates the firmware to v1.7. So which firmware do I need to get?!

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

janjongboom, 2019-03-13:
Could it be that NUCLEO-F446RE is dropping bytes on 115,200 baud rate? I think I heard something earlier about that.

janjongboom, 2019-03-13:
Updated to Mbed OS 5.12 this problem seems to go away. Did we fix anything in flow control handling between 5.11 and 5.12?

cmonr, 2019-03-13:
@ARMmbed/mbed-os-ipcore @kjbracey-arm 

I recall seeing a bunch of fixes come in for the ESP32 recently.

SeppoTakalo, 2019-03-14:
@cmonr Arm is not providing ESP32 driver, this is ESP8266 related.

The correct firmware for ESP8266 should be v1.7.

@janjongboom if the driver fails to parse the AT firmware version, it might be that it gets corrupted data from serial port.

We don't have that particular board here, so I cannot directly verify.
Anyway, this bug should be assigned to IP Core team for fixing.

Jan, can you try enabling the AT debugging on that driver
https://github.com/ARMmbed/mbed-os/blob/master/components/wifi/esp8266-driver/ESP8266Interface.h#L65

it should print out the commands and responses from serial port.

cmonr, 2019-03-15:
> @cmonr Arm is not providing ESP32 driver, this is ESP8266 related.

Whoops, my mistake.

janjongboom, 2019-03-15:
@SeppoTakalo Yeah, I have. I think there's some flow control fixes in master perhaps? Because I don't see this issue on 5.12.

closed 2019-03-15 by janjongboom