ARMmbed/mbed-os #10085: F446RE hard faults when enabling MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER

janjongboom, 2019-03-13:
### Description

I'm running [http-example](https://os.mbed.com/teams/sandbox/code/http-example/) on NUCLEO-F446RE with ESP8266 hooked up. Running Mbed OS 5.12 (c0939781ea1fa50657e055867193b62b1c9a6035). When setting both tickless and `MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER=0` the board immediately hard faults with:

```
++ MbedOS Error Info ++
Error Status: 0x80010131 Code: 305 Module: 1
Error Message: Kernel Error: 0xFFFFFFFF, Unspecified RTOS error
Location: 0x8008B55
Error Value: 0xFFFFFFFF
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : FFFFFFF8
R1   : 00B2D920
R2   : FFFFFFFF
R3   : 00000000
R4   : 00000000
R5   : 2001FF68
R6   : 00B2D920
R7   : 00000000
R8   : 2001FDD8
R9   : 2001FDD0
R10  : 0801B7B0
R11  : 00000000
R12  : FFFFFFFF
SP   : 2001FCF8
LR   : 080131B3
PC   : 08010CE4
xPSR : 610F000B
PSP  : 00000000
MSP  : 2001FC90
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000008
AFSR : 00000000
BFAR : 00B2D920
Mode : Handler
Priv : Privileged
Stack: MSP

-- MbedOS Fault Handler --
```

```
Crash Info:
	Crash location = strlen [0x08010CE4] (based on PC value)
	Caller location = _svfprintf_r [0x080131B3] (based on LR value)
	Stack Pointer at the time of crash = [2001FCF8]
	Target and Fault Info:
		Processor Arch: ARM-V7M or above
		Processor Variant: C24
		Forced exception, a fault with configurable priority has been escalated to HardFault
		A precise data access error has occurred. Faulting address: 00B2D920
```

I have made the idle thread stack size bigger already, this is my mbed_app.json:

```json
{
    "config": {
        "main-stack-size": {
            "value": 8192
        }
    },
    "macros": [
        "MBED_TICKLESS=1",
        "MBED_CPU_STATS_ENABLED=1",
        "MBED_SLEEP_TRACING_ENABLED=1",
        "MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER=0",
        "MBED_CONF_APP_IDLE_THREAD_STACK_SIZE=2048",
        "MBEDTLS_MPI_MAX_SIZE=1024",
        "MBEDTLS_MPI_WINDOW_SIZE=1",
        "MBEDTLS_USER_CONFIG_FILE=\"mbedtls_entropy_config.h\"",
        "MBEDTLS_TEST_NULL_ENTROPY",
        "MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES",
        "MBED_HEAP_STATS_ENABLED=1"
    ],
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200,
            "platform.stdio-convert-newlines": true,
            "mbed-mesh-api.6lowpan-nd-channel-page": 0,
            "mbed-mesh-api.6lowpan-nd-channel": 12,
            "mbed-trace.enable": null,
            "mbed-http.http-buffer-size": 2048,
            "nsapi.default-wifi-security": "WPA_WPA2",
            "nsapi.default-wifi-ssid": "\"SSID\"",
            "nsapi.default-wifi-password": "\"Password\""
        },
        "DISCO_L475VG_IOT01A": {
            "target.network-default-interface-type" : "WIFI"
        },
        "NUCLEO_F446RE": {
            "target.macros_add": [
                "MBED_CONF_TARGET_STDIO_UART_TX=D8",
                "MBED_CONF_TARGET_STDIO_UART_RX=D2"
            ]
        }
    }
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

MarceloSalazar, 2020-07-28:
@ARMmbed/team-st-mcd please review and let us know what you think

jeromecoutant, 2020-08-12:
> @ARMmbed/team-st-mcd please review and let us know what you think

Hi
I think NUCLEO-F446RE doesn't support TICKLESS...
So we should close the issue
Regards,

0xc0170, 2020-08-12:
Closing as not supported.

closed 2020-08-12 by 0xc0170