ARMmbed/mbed-os #10090: Error is too verbose when a compiler executable is not found

bridadan, 2019-03-13:
### Description

If you do not have a compiler executable's path correctly configured, the current error is quite verbose:

```
$ mbed compile -t ARM -m UBLOX_C030_U201
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-os" (program)
usage: make.py [-h] [-m MCU] [--custom-targets CUSTOM_TARGETS_DIRECTORY]
               [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG]
               [-p PROGRAM | -n PROGRAM | -L | -S [{matrix,toolchains,targets}]]
               [-j JOBS] [-v] [--silent] [-D MACROS] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [--ignore IGNORE] [-b BAUD] [--rpc] [--usb] [--dsp] [--testlib]
               [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: Could not find executable for ARMC6.
Currently set search path: No path set
[mbed] ERROR: "C:\Users\bridan01\AppData\Local\Programs\Python\Python36-32\python.exe" returned error.
       Code: 2
       Path: "C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-os"
       Command: "C:\Users\bridan01\AppData\Local\Programs\Python\Python36-32\python.exe -u C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-os\tools\make.py -t ARM -m UBLOX_C030_U201 --source . --build .\BUILD\UBLOX_C030_U201\ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

It's hard to find the actual problem in all the output. It's also misleading since the fault isn't with the arguments supplied, but rather with your environment. A more appropriate error would be similar to this:

```
$ mbed compile -m K64F -t ARM
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-os" (program)
Could not find executable for ARMC6.
Currently set search path: No path set
[mbed] ERROR: "C:\Users\bridan01\AppData\Local\Programs\Python\Python36-32\python.exe" returned error.
       Code: 1
       Path: "C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-os"
       Command: "C:\Users\bridan01\AppData\Local\Programs\Python\Python36-32\python.exe -u C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\BUILD\K64F\ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

This error is generated by a call to the `args_error` function in our tools: https://github.com/ARMmbed/mbed-os/blob/master/tools/utils.py#L330

It seems unnecessary to always print all the parser options. Usually the actual error gets drowned out by all the optional arguments we have in our scripts. We can probably skip the printing of the parser options in general for this function.

FYI @theotherjimmy 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

theotherjimmy, 2019-03-13:
I'm pretty sure that almost every args related error is similarly too verbose. See what happens with `mbed compile -t FOOBAR -m K64F` or similar. I think we should reserve the usage information for when a user specifies an argument that we don't accept. so `mbed compile --foo` should print usage, but missing compilers and incorrect toolchain arguments should not.

bridadan, 2019-03-25:
This was fixed in #10193 .

closed 2019-03-25 by bridadan