ARMmbed/mbed-os #10095: Incorrect behavior working with Crypto HW accelerator on STM32F439xI chipset

avolinski, 2019-03-14:
### Description
Incorrect behavior working with Crypto HW accelerator on STM32F439xI chipset.
initially detected on UBLOX_EVK_ODIN_W2 board.
Running compliance test suite fails in test037 using PSA_ALG_CBC_PKCS7 algorithm
Attestation test also fails when SHA in HW is enabled.

Toolchains: GCC ARM, IAR8
Target platform: UBLOX_EVK_ODIN_W2 

Steps to reproduce:

- git clone git@github.com:ARMmbed/mbed-os-psa-compliance-tests-example.git
- cd mbed-os-psa-compliance-tests-example
- mbed deploy

Compile and run latest mbedOS with the following command:
- mbed test -m UBLOX_EVK_ODIN_W2 -t GCC_ARM -n *psa*compliance* -DMBEDTLS_USER_CONFIG_FILE=\"../comp_config.h\" -v

Expected result:
components-target_psa-tests-compliance_crypto-test_c037 test should FAIL
components-target_psa-tests-compliance_attestation-test_a001 test should FAIL

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-03-14:
cc @ARMmbed/team-st-mcd @ARMmbed/team-ublox

jeromecoutant, 2020-04-02:
Hi @ARMmbed/team-ublox

Is it possible to check #12747 (#10088 has been reverted) ?

Thx

aqib-ublox, 2020-04-07:
@jeromecoutant  Yes we verified solution working on master branch and its working but we have paused ODIN_W2 migration to Mbed6. @MarceloSalazar can u find feasibility of supporting it for mbed 5.15 as well because we are getting merge conflicts with 5.15. 
cc @qasim-ublox

0xc0170, 2020-04-07:
what merge conflicts? 5.15 is a different base. In case something from PR 12747 is needed in 5.15, it needs to be backported (cherry picked to 5.15 branch) - conflicts might appear due to the diverged branches.

jeromecoutant, 2020-06-16:
@0xc0170 @MarceloSalazar 
Please close this issue as #12747 is now merged.

In mbed-os-5.15 branch, you have to backport #12626 first before #12747

closed 2020-10-15 by adbridge