ARMmbed/mbed-os #10103: uvision project generated does not have correct memory information

mmahadevan108, 2019-03-14:
### Description
The uvision project generated by the mbed exporter does not have the memory information provided inside index.json

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

bridadan, 2019-03-15:
Thanks for raising the issue @mmahadevan108, I believe I have part of the fix in this PR: https://github.com/ARMmbed/mbed-os/pull/10045. This fixes the issue of the RAM start and size being incorrect.

@theotherjimmy is also working on a fix in cmsis-pack-manager: https://github.com/ARMmbed/cmsis-pack-manager/pull/79. That should fix the problem of no flashing algorithm being set.

theotherjimmy, 2019-03-15:
@bridadan Doing a release. I'm hoping I don't run into any python3 incompatibilities this time.

adbridge, 2020-10-15:
I'm closing this as we believe it is fixed. If this is not the case please raise a new issue.

closed 2020-10-15 by adbridge