ARMmbed/mbed-os #10107: OS2 ARM build is no more possible

jeromecoutant, 2019-03-14:
### Description

I couldn't compile targets with ARM.

Ex:
python ./tools/build.py -m NUCLEO_L476RG -t ARM
The target NUCLEO_L476RG does not support the toolchain ARMC6
NUCLEO_L476RG supports ARM, uARM, IAR, GCC_ARM toolchains

python ./tools/build.py -m NUCLEO_L476RG -t uARM
The target NUCLEO_L476RG does not support the toolchain ARMC6
NUCLEO_L476RG supports ARM, uARM, IAR, GCC_ARM toolchains

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2019-03-15:
@ARMmbed/mbed-os-tools @SenRamakri

0xc0170, 2019-03-15:
I could reproduce this error as well on master today.

@theotherjimmy will mbed 2 builds be broken in the online compiler?

theotherjimmy, 2019-03-15:
@0xc0170 No

theotherjimmy, 2019-04-08:
@jeromecoutant This should be resolved. The new error message:
```bash
> python ./tools/build.py -m NUCLEO_L476RG -t ARM
------------------------------------------------------------
Warning: We noticed that you are using Arm Compiler 5. We are deprecating the use of Arm Compiler 
5 soon. Please upgrade your environment to Arm Compiler 6 which is free to use with Mbed OS. For 
more information, please visit https://os.mbed.com/docs/mbed-os/latest/tools/index.html
------------------------------------------------------------
Could not find executable for ARMC5.
Currently set search path: No path set
```

theotherjimmy, 2019-04-08:
I spoke too soon, seems uARM is still buggy WRT toolchain detection logic.

PR #10347 (2019-04-10): Fix detection of supported targets for mbed 2 builds
### Description

Fix #10107.

This should get Mbed 2 builds going again with ARMC6. The uARM logic doesn't seem to be 100%, but honestly I'm not really sure what the expected behavior is for it.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes

@jeromecoutant
closed 2019-04-10 by 0xc0170