ARMmbed/mbed-os #10109: map file extraction is failing with GCC and M33

jeromecoutant, 2019-03-14:
### Description

I am compiling M33 targets in GCC.

Binary is OK, but
map file analysing is failing, it gives RAM and FLASH size to 0:

Total Static RAM memory (data + bss): 0(+0) bytes
Total Flash memory (text + data): 0(+0) bytes

map.html file is also empty

Note also that long name makes also some warnings:

````
.tfm_bl2_shared_data
                0x30010000      0x400 load address 0x0c03e200
.TFM_SECURE_STACK
                0x30011800     0x2000 load address 0x0c03e200
.TFM_UNPRIV_SCRATCH
                0x30013800      0x400 load address 0x0c03e200
````

````
Unknown object name found in GCC map file: load address 0x0c03e200
Unknown object name found in GCC map file: load address 0x0c03e200
Unknown object name found in GCC map file: load address 0x0c03e200
````

@theotherjimmy 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-03-14:
@ARMmbed/mbed-os-psa

jeromecoutant, 2019-03-22:
@theotherjimmy this script is very useful for new development with M33.
Thx

theotherjimmy, 2019-04-08:
@jeromecoutant I just tried with the `LPC55S69_NS`, an M33 on master, and the map file was parsed correctly. Which target are you using?

theotherjimmy, 2019-04-08:
Alternatively, you can post the map file, and I'll modify `memap` to parse that correctly.

theotherjimmy, 2019-04-08:
Update: Jerome sent me a reproducing map file offline. Section parsing is not working correctly, and all sections are marked as `unknown` which is causing a table of `0`s

PR #10348 (2019-04-10): Parse M33 + GCC_ARM map files
### Description

Turns out that under certain circumstances, `GCC_ARM` will emit
whitespace-delimited text in map files. Before this PR, Memap's
`GCC_ARM` map file parsing logic assumes that sections assumes
that the sections always start the string. This PR allows the
sections to be prefixed by whitespace.

Fixes #10109

Note for reviewers: the first commit cleans up lint warnings and
the second commit fixes the bug.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
     [... cut]
closed 2019-04-10 by cmonr