ARMmbed/mbed-os #10111: [mbed os]: Greentea BLE testcase is failing with "timeout" error

UrmilaKL, 2019-03-14:
mbed- cli version: 1.8.3
mbed os : 5.11.4

Description: [mbed os]: Greentea test case is failing with "timeout" error

Steps: 1. Fetch latest repo from the git server
2. Create folder structure as below under TESTS folder
TESTS/Group1/TestCase1/main.cpp

Find the below main.cpp attached, Also I have attached the logs with -vv option

Note: I added the 'printf' statement in each of the function where it is getting stuck in "dispatch forever() method, If we remove that code of line test case is passing
Let me know if anything else I need to add here

C:\mbed-examples\mbed-os>mbed test -m CY8CKIT_062_WIFI_BT -v -n tests-Group1-setADV_param -vv
[mbed-7200] Working path "C:\mbed-examples\mbed-os" (program)
[mbed-7200] Exec "c:\python27\python.exe -m pip list -l" in "C:\mbed-examples\mbed-os"
Package                       Version
----------------------------- -----------
appdirs                       1.4.3
argcomplete                   1.9.4
asn1ate                       0.6.0
asn1crypto                    0.24.0
beautifulsoup4                4.1.0
certifi                       2018.11.29
cffi                          1.11.5
chardet                       3.0.4
colorama                      0.4.1
cryptography                  2.1.4
Deprecated                    1.2.4
ecdsa                         0.13
enum34                        1.1.6
fasteners                     0.14.1
funcsigs                      1.0.2
functools32                   3.2.3.post2
future                        0.17.1
futures                       3.2.0
fuzzywuzzy                    0.11.0
hgapi                         1.7.4
icetea                        1.0.2
idna                          2.7
intelhex                      2.2.1
intervaltree                  3.0.2
ipaddress                     1.0.22
Jinja2                        2.7.3
jsonmerge                     1.5.2
jsonpointer                   1.14
jsonschema                    2.6.0
junit-xml                     1.8
lockfile                      0.12.2
Logbook                       1.4.1
lxml                          4.2.5
manifest-tool                 1.4.6
MarkupSafe                    1.1.0
mbed-cli                      1.8.3
mbed-cloud-sdk                2.0.1
mbed-flasher                  0.9.2
mbed-greentea                 1.5.0
mbed-host-tests               1.4.4
mbed-ls                       1.7.1
mbed-os-tools                 0.0.2
mbed-test-wrapper             1.0.0
mock                          2.0.0
monotonic                     1.5
nmap                          0.0.1
ntfsutils                     0.1.4
pathlib                       1.0.1
pbr                           5.1.1
pip                           19.0.1
prettytable                   0.7.2
[... 201 lines cut ...]
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test on hardware with target id: 190000006c170b07036c170b00000000000000002e127069
mbedgt: test suite 'tests-group1-setadv_param' ....................................................... TIMEOUT in 145.62 sec
        test case: 'Testing success test' ............................................................ ERROR in 0.00 sec
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.9513527867
mbedgt: test suite report:
| target                      | platform_name       | test suite                | result  | elapsed_time (sec) | copy_method |
|-----------------------------|---------------------|---------------------------|---------|--------------------|-------------|
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-group1-setadv_param | TIMEOUT | 145.62             | default     |
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
| target                      | platform_name       | test suite                | test case            | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|---------------------------|----------------------|--------|--------|--------|--------------------|
| CY8CKIT_062_WIFI_BT-GCC_ARM | CY8CKIT_062_WIFI_BT | tests-group1-setadv_param | Testing success test | 0      | 0      | ERROR  | 0.0                |
mbedgt: test case results: 1 ERROR
mbedgt: completed in 150.01 sec
mbedgt: exited with code 1
[mbed-7200] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\mbed-examples\mbed-os"
       Command: "mbedgt --test-spec .\BUILD\tests\CY8CKIT_062_WIFI_BT\GCC_ARM\test_spec.json -n tests-Group1-setADV_param -V"
       Tip: You could retry the last command with "-v" flag for verbose output
---

UrmilaKL, 2019-03-14:
Please find the test case here,
/* mbed Microcontroller Library
 * Copyright (c) 2006-2015 ARM Limited
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#include <events/mbed_events.h>
#include <mbed.h>
#include "ble/BLE.h"
#include "ble/Gap.h"
#include "ble/services/HeartRateService.h"
#include "greentea-client/test_env.h"
#include "utest.h"
#include "rtos.h"
#include "unity.h"

using namespace utest::v1;

void schedule_ble_events(BLE::OnEventsToProcessCallbackContext *context);
const static char DEVICE_NAME[] = "MBEDOSHeartrate";

static events::EventQueue event_queue(/* event count */ 16 * EVENTS_EVENT_SIZE);

class HeartrateDemo : ble::Gap::EventHandler {
public:
    HeartrateDemo(BLE &ble, events::EventQueue &event_queue) :
        _ble(ble),
        _event_queue(event_queue),
        _led1(LED1, 1),
        _connected(false),
        _hr_uuid(GattService::UUID_HEART_RATE_SERVICE),
        _hr_counter(100),
        _hr_service(ble, _hr_counter, HeartRateService::LOCATION_FINGER),
        _adv_data_builder(_adv_buffer) { }

    void start() {
		printf("I am in start\n");
        _ble.gap().setEventHandler(this);
         printf("SetEventHandler called\n");
        _ble.init(this, &HeartrateDemo::on_init_complete);
         printf("Initialization Completed\n");
        _event_queue.call_every(500, this, &HeartrateDemo::blink);
		printf("Blink called\n");
        _event_queue.call_every(1000, this, &HeartrateDemo::update_sensor_value);
           printf("Updated sensor called\n");
       _event_queue.dispatch_forever();
		printf("dispatch called\n");
    }

private:
    /** Callback triggered when the ble initialization process has finished */
    void on_init_complete(BLE::InitializationCompleteCallbackContext *params) {
        if (params->error != BLE_ERROR_NONE) {
            printf("Ble initialization failed.");
            return;
        }

       // print_mac_address();

[... 114 lines cut ...]
	 
}

void schedule_ble_events(BLE::OnEventsToProcessCallbackContext *context) {
    event_queue.call(Callback<void()>(&context->ble, &BLE::processEvents));
}

utest::v1::status_t test_setup(const size_t number_of_cases) {
    // Setup Greentea using a reasonable timeout in seconds
    GREENTEA_SETUP(100, "default_auto");
    return verbose_test_setup_handler(number_of_cases);
}

Case cases[] = {
	
    Case("Testing success test", display)
};

Specification specification(test_setup, cases);

int main()
{
	printf("I am in main()");
   return !Harness::run(specification);
}

UrmilaKL, 2019-03-15:
Is there any method so that we can pass time in dispatch_forever() function?
like below
dispatch(100);

0xc0170, 2019-03-15:
cc @ARMmbed/mbed-os-pan

desmond-blue, 2019-03-18:
You can pass in the timeout for dispatch events, please check the [documentation](https://os.mbed.com/docs/mbed-os/v5.11/mbed-os-api-doxy/classevents_1_1_event_queue.html#a397fcb417425ebaa28c20b9e90ef3ed5).

Take your code for example, `dispatch(100)` means "Executes events until 100 milliseconds have passed".

linlingao, 2019-03-19:
@UrmilaKL I believe @desmond-blue has answered your question. Please close this issue if you have no further questions.

UrmilaKL, 2019-03-21:
HI @Desmond-blue,
If I add dispatch(100) to my code instead of dispatch_forever() test case is passing,
But 
Like above if I try to add one more class with one more display1 test case, My test case is failing with "TIMEOUT" error, First test case will pass but second one will fail
What may be the reason?
Thanks
Urmila Bhat

desmond-blue, 2019-03-26:
@UrmilaKL You could try destructing the BLE and HeardRate objects, after the test case 1 finished.

And please post this kind of questions on our [Question](https://os.mbed.com/questions/) or [Forum](https://os.mbed.com/forum/) because the two places are best fitted for discussing the application. We really appreciate that.

linlingao, 2019-05-30:
It appears Desmond has already answered your question.

closed 2019-05-30 by linlingao