ARMmbed/mbed-os #10112: Timeout when running  IceTea test on CY8CKIT_062_WIFI_BT

ShuopengDeng, 2019-03-14:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X ] Bug

ShuopengDeng, 2019-03-14:
The auto formatter somehow deleted my description.

It is a timing issue during startup.
There are 2 ways to run icetea.

The first way is through mbed by calling mbed test -m CY8CPROTO_062_4343W -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED

This way compiles, flashs the application, and drives the icetea test. The fails because the test start driving the icetea test before the device's uart is ready. The test fails due to timeout waiting for a response to a command that the device never received.

 

The second way is by manually programming the application and invoking icetea directly by calling icetea --tc test_cmdline

This assume that the application has already been flash and just drives the test. The test passes in this case.

 

The only way I can find to fix the timing issue is to modify the mbed-os\TEST_APPS\testcases\example\test_cmdline.py file to add an ready trigger.

"application": {
   "name": "TEST_APPS-device-exampleapp",
   "cli_ready_trigger": "ARM Ltd"
}

This wait cause the test to wait until it receives the text "ARM Ltd" from UART before it start driving the test. This fixing the case number 1 but breaks case number 2. In case number 2, the application was already flashed so icetea command fails during initialization waiting for a command sent long ago.

cmonr, 2019-03-15:
@ShuopengDeng The original comment was made within a comment block. Fixed it for you.

jupe, 2019-03-17:
Another, perhaps easier way is to use `--sync_start` -option when starting icetea. [Here](https://github.com/ARMmbed/icetea/blob/master/doc/Icetea.md#command-line-parameters) is more details about this option. 

Seems that this same issue is quite common problem so maybe `--sync_start` could be activated by default at `mbed-cli`. I think it should not cause any side effects in these tests unless some combination of mbed board&SW stuck if it receive serial data before it's really ready to receive... @OPpuolitaival, @bridadan please consider this.

bridadan, 2019-03-18:
@jupe I didn't see any details about `--sync_start` in the document you linked. But if you'd like change the arguments icetea is invoked with from Mbed CLI, they can be added here: https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L2876-L2879

jupe, 2019-03-19:
oh, true. Its [not yet](https://github.com/ARMmbed/icetea/pull/85) released.

0xc0170, 2019-08-14:
Version 2 was released. This should be resolved

closed 2019-08-14 by 0xc0170