ARMmbed/mbed-os #10119: GR_LYCHEE: greentea ticker test case failures

VeijoPesonen, 2019-03-15:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-03-15:
only GCC , other toolchains not affected?

VeijoPesonen, 2019-03-15:
@0xc0170 I haven't tried with other toolchains.

d-kato, 2019-03-25:
@VeijoPesonen, It was not possible to build with https://github.com/ARMmbed/mbed-os/commit/6906f39bf586aaafac1d0a23eadb604c0905c476.
```
Traceback (most recent call last):
  File "C:\xxxx\mbed-os\tools\test.py", line 29, in <module>
    from tools.config import ConfigException, Config
  File "C:\xxxx\mbed-os\tools\config\__init__.py", line 37, in <module>
    from ..arm_pack_manager import Cache
  File "C:\xxxx\mbed-os\tools\arm_pack_manager\__init__.py", line 22, in <module>
    from cmsis_pack_manager import Cache as _Cache
ImportError: No module named cmsis_pack_manager
```
So I tried with the latest revision https://github.com/ARMmbed/mbed-os/commit/b29b55ab5a9ec364d83f3d3ac175c8ac160ae2e2 on the master.

The result was successful except for the newly added ``mbed-os-tests-mbed_platform-wait_ns``.
```
+-------------------+---------------+--------------------------------------------+--------+--------------------+-------------+
| target            | platform_name | test suite                                 | result | elapsed_time (sec) | copy_method |
+-------------------+---------------+--------------------------------------------+--------+--------------------+-------------+
| GR_LYCHEE-GCC_ARM | GR_LYCHEE     | mbed-os-tests-mbed_drivers-ticker          | OK     | 44.9               | default     |
| GR_LYCHEE-GCC_ARM | GR_LYCHEE     | mbed-os-tests-mbed_drivers-timeout         | OK     | 58.21              | default     |
| GR_LYCHEE-GCC_ARM | GR_LYCHEE     | mbed-os-tests-mbed_hal-common_tickers_freq | OK     | 38.39              | default     |
| GR_LYCHEE-GCC_ARM | GR_LYCHEE     | mbed-os-tests-mbed_platform-wait_ns        | FAIL   | 14.36              | default     |
| GR_LYCHEE-GCC_ARM | GR_LYCHEE     | mbed-os-tests-mbedmicro-rtos-mbed-basic    | OK     | 23.68              | default     |
+-------------------+---------------+--------------------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 FAIL / 4 OK
```
We need to fix ``mbed-os-tests-mbed_platform-wait_ns``, so I will pull request it.

VeijoPesonen, 2019-03-25:
> 
> 
> @VeijoPesonen, It was not possible to build with [6906f39](https://github.com/ARMmbed/mbed-os/commit/6906f39bf586aaafac1d0a23eadb604c0905c476).
> 
> ```
> Traceback (most recent call last):
>   File "C:\xxxx\mbed-os\tools\test.py", line 29, in <module>
>     from tools.config import ConfigException, Config
>   File "C:\xxxx\mbed-os\tools\config\__init__.py", line 37, in <module>
>     from ..arm_pack_manager import Cache
>   File "C:\xxxx\mbed-os\tools\arm_pack_manager\__init__.py", line 22, in <module>
>     from cmsis_pack_manager import Cache as _Cache
> ImportError: No module named cmsis_pack_manager
> ```
```
cd mbed-os
pip install -r requirements.txt
```
That should solve the issue you're having

VeijoPesonen, 2019-03-25:
@d-kato thanks, once you create the PR I'll give it a try.

d-kato, 2019-03-25:
@VeijoPesonen, I created the PR. Please confirm.

VeijoPesonen, 2019-03-28:
#10213  look good to me.

closed 2019-03-28 by VeijoPesonen