ARMmbed/mbed-os #10126: NRF52840 returning completely different 16 bit ADC values for the same voltage.

yogeshk19, 2019-03-15:
Description
- Target : NRF52840
- Toolchain : GCC_ARM
- mbed-cli
- Mbed OS 5.11.5
We currently have a custom board that houses a NRF52840 BLE module. 

1) One application we wrote just to test reading the voltage at pin P0_28, by configuring the pin as Analog input and we are reading using the Analog Input API and we are reading 16 bit value.  The application just reads these voltage every 1 minute, but samples 20 readings every 10 ms and we take an average of those 20 readings.

2) The second application does what the above application does, but is a more complete application, which operates as a BLE central device, where in it scans for BLE devices if need be and also reads serial data over RS-232 in 1 minute intervals and reads the pin P0_28 at the same intervals as in the simpler application called out earlier.

The voltage to be read after applying the necessary conversions should be 12.3v. However both applications return completely different ADC values for the same voltage of 12.3 v.  The average of 20 ADC 16 bit readings for the first application is 9505 and the seconds application is 7755. 

So I am wondering if this is a bug or does the ADC need to be configured differently in NRF52840? I am not sure how two different applications reading from P0_28 on the same custom board would return two different set of ADC values? 

Also the board is powered by an external power supply.

Issue request type
[ ] Question
[ ] Enhancement
[X] Bug

Thanks,
Yogesh

TacoGrandeTX, 2019-03-29:
@yogeshk19 Sorry for not replying sooner but I was out for a week and still getting caught up.  Future responses will likely be delayed. 

> So I am wondering if this is a bug or does the ADC need to be configured differently in NRF52840? 

We're not sure what you mean by "differently".  Can you explain the context?  There is a default configuration for the ADC and channel, but it might not fit your use case.  Are you using the default analog_init() or have you made modifications?    

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/analogin_api.c#L43-L102

> The average of 20 ADC 16 bit readings for the first application is 9505 and the seconds application is 7755.

Have you checked the 20 values from the BLE app?  Are there a few oddball samples or are they all lower?  What is the range of values you read in the basic app?

yogeshk19, 2019-04-08:
@TacoGrandeTX I have not made any changes to the analogin_api.c. We are currently using pin P0.28 as analogin input. When no firmware is loaded or rather the chip erased, the voltage measured at the pin is 0.435v, as per the multi-meter and that should be the right voltage we should be reading. However when we install the application firmware, the voltage at this pin measured by the multi-meter drops to 0.235v. Since this is a custom board, the board manufacturer's believe that GPIO pins are being configured with a pull down resistor and no where in our code base are we configuring any pins to be pulled down. Today under issues I saw an issue around cordio ble stack has some unwanted GPIO control, #10321 , could this issue be causing the voltage drop?

desmond-blue, 2019-07-31:
Hi @yogeshk19, sorry for getting this lately, do you need further help on this issue? Thanks.

desmond-blue, 2019-08-26:
@0xc0170 I guess we can close it since no feedback from reporter for almost one month?

0xc0170, 2019-08-26:
Yes, there has been an update to nordic and others. If there is still an issue, this needs to be updated

closed 2019-08-26 by 0xc0170