ARMmbed/mbed-os #10132: [mbed] ERROR: "/usr/bin/python" returned error

ccu1tn, 2019-03-17:
Dear everyone.
I install tools and compiler the ARM program by this command "mbed compile". It runs 100% but has some errors. 
[image]

I don't know how to fix it? who can anyone share for me so way to fix it?
Thank you so much.

ccu1tn, 2019-03-18:
Dear @ciarmcom 
This link is dead. I can't access it.

0xc0170, 2019-03-18:
@ccu1tn It's internal jira reference created by our bot.

what do you compile? The linker error - can't find main - missing `main()` function

ccu1tn, 2019-03-18:
Hi @0xc0170 ;
Thank for your help. It is miss correct. This is my file I need to compile with mbed. 
[image]
when I compile by the command "mbed compile" . It has errors " [Fatal Error] main.cpp@15,46: arm_math.h: No such file or directory". it means not find arm_math.h in the folder.
Now, I want to call automatic to that arm_math.h library, how can I do it? 
Please give me some of your idea
Many thank

0xc0170, 2019-03-18:
Where is `arm_math.h` located? It's nor part of Mbed OS. Without much details about your environment, we can only guess.

Based on your main above, arm_math is not needed, to test blinky.

ccu1tn, 2019-03-18:
Dear @0xc0170 ;

I know this program don't need that library. but I have a program need that library, 
Can you tutorial me config mbed compie ? Maybe I configured mbed compile wrong.

0xc0170, 2019-03-18:
Understood. My point was, without having more details, we can't help you much. Where is `arm_mat.h` located? Can you share the tree? Can you share the console output using verbose? I would like to see your include paths (if  arm_math.h location is included or not).

ccu1tn, 2019-03-18:
The arm_math.h inside here 
[image]

0xc0170, 2019-03-19:
We are closing this, please continue in the mbed-cli repo, this is duplicate

Note, please add there details about your structure. Also use -v to get include paths and verify all is getting included. Would also help to print the tree structure in your blinky.

closed 2019-03-19 by 0xc0170