ARMmbed/mbed-os #10134: [OOB 5.12.0-rc2 CoreOS]Blinky got failed compiling with ARM toolchain

soramame21, 2019-03-18:
### Description
Blinky got failed compiling with ARM toolchain.
GCC_ARM and IAR can build the example without error.
### Example to build
https://os.mbed.com/teams/ST/code/Nucleo_blink_led/
### Steps
```
mbed import https://edamame22@os.mbed.com/teams/ST/code/Nucleo_blink_led/
cd Nucleo_blink_led
mbed remove mbed
mbed add mbed-os
cd mbed-os
git checkout -b ren01 mbed-os-5.12.0-rc2
cd ../
C:\myTest\oob\Nucleo_blink_led>armcc --help
Product: MDK Professional 5.22
Component: ARM Compiler 5.06 update 4 (build 422)
Tool: armcc [4d3604]

C:\myTest\oob\Nucleo_blink_led>mbed compile -m K64F -t ARM -v
[mbed] Working path "C:\myTest\oob\Nucleo_blink_led" (program)
[mbed] Exec "c:\python27\python.exe -u C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\BUILD\K64F\ARM -v" in "C:\myTest\oob\Nucleo_blink_led"
[OS ERROR] Command: armclang --vsn
Traceback (most recent call last):
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\make.py", line 74, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\build_api.py", line 452, in build_project
    toolchain.version_check()
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\toolchains\arm.py", line 107, in version_check
    stdout, _, retcode = run_cmd([self.cc[0], "--vsn"], redirect=True)
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\utils.py", line 108, in run_cmd
    universal_newlines=True)
  File "c:\python27\lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "c:\python27\lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\myTest\oob\Nucleo_blink_led"
       Command: "c:\python27\python.exe -u C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\BUILD\K64F\ARM -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

ashok-rao, 2019-03-18:
@screamerbg ..FYI.

linlingao, 2019-03-18:
@soramame21 The default ARM compiler has switched to AC6. The compiler used is armclang instead of armcc. Please verify you have AC6 installed and armclang is in your PATH.

studavekar, 2019-03-18:
Hi,

as @linlingao suggested we need to install AC6, below are the steps 

1. AC6.11 can be downloaded from https://developer.arm.com/products/software-development-tools/compilers/arm-compiler/downloads/version-6 
2. Install AC6, by default installation location C:/Program Files/ARMCompiler6.11/bin
3. configure the path AC6 path `mbed config MBED_ARMC6 "C:/Program Files/ARMCompiler6.11/bin"`  please test this with 5.12.0-rc3. 
4. run compile `mbed compile -m K64F -t ARM -v`

cmonr, 2019-03-18:
@studavekar @AnotherButler There are OOB docs that instruct how to do this, correct?

Sounds like this would be a good way to test them out.

soramame21, 2019-03-19:
Where can I set a license for AC6? or should I ask a separate license for it?

cmonr, 2019-03-19:
@soramame21 I've emailed you instructions on OoB testing, since the link is private and am not sure you got them.

soramame21, 2019-03-19:
@cmonr Thank you for the email. the floating license is working now. but I got a new error and cannot find  5.12.0-rc3, it seems not be published yet.

```
C:\myTest\oob\Nucleo_blink_led>mbed compile -m K64F -t ARM -v -c
[mbed] Working path "C:\myTest\oob\Nucleo_blink_led" (program)
[mbed] Exec "c:\python27\python.exe -u C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\BUILD\K64F\ARM -c -v" in "C:\myTest\oob\Nucleo_blink_led"
Building project Nucleo_blink_led (K64F, ARM)
Scan: Nucleo_blink_led
Macros: -DTARGET_PSA -DCOMPONENT_SD=1 -DDEVICE_EMAC=1 -D__MBED__=1 -DDEVICE_USBDEVICE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_NAME=K64F -DCOMPONENT_FLASHIAP=1 -DDEVICE_RTC=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_ARMC6 -DMBED_BUILD_TIMESTAMP=1552962705.75 -DFSL_RTOS_MBED -DTARGET_KPSDK_MCUS -DDEVICE_USTICKER=1 -DDEVICE_CRC=1 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_K64F -DCOMPONENT_PSA_SRV_IMPL=1 -DDEVICE_STORAGE=1 -DDEVICE_LPTICKER=1 -DDEVICE_PWMOUT=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_Freescale_EMAC -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_I2CSLAVE=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DCOMPONENT_NSPE=1 -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -DMBEDTLS_PSA_CRYPTO_C -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DTARGET_MCUXpresso_MCUS -DDEVICE_SPI=1 -DTOOLCHAIN_ARM_STD -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_PORTIN=1 -DTOOLCHAIN_ARM -DTARGET_MCU_K64F -DARM_MATH_CM4
Compile [  0.1%]: except.S
Compile: armasm --cpu=Cortex-M4.fp --cpreproc --cpreproc_opts=-c,--target=arm-arm-none-eabi,-mthumb,-Os,-Wno-armcc-pragma-push-pop,-Wno-armcc-pragma-anon-unions,-DMULADDC_CANNOT_USE_R7,-fdata-sections,-fno-exceptions,-MMD,-D_LIBCPP_EXTERN_TEMPLATE(...)=,-fshort-enums,-fshort-wchar,-DMBED_TRAP_ERRORS_ENABLED=1,-mcpu=cortex-m4,-mfpu=fpv4-sp-d16,-mfloat-abi=hard,-DMBED_ROM_START=0x0,-DMBED_ROM_SIZE=0x100000,-DMBED_RAM_START=0x20000000,-DMBED_RAM_SIZE=0x30000,-DMBED_RAM1_START=0x1fff0000,-DMBED_RAM1_SIZE=0x10000,-D__CORTEX_M4,-DCPU_MK64FN1M0VMD12,-DARM_MATH_CM4,-D__FPU_PRESENT=1,-D__MBED_CMSIS_RTOS_CM,-D__CMSIS_RTOS,-DFSL_RTOS_MBED,-DMBEDTLS_PSA_CRYPTO_C,@.\BUILD\K64F\ARM\.includes_eb3287dba475f46d14e0481221c0d1c9.txt,-include,.\BUILD\K64F\ARM\mbed_config.h -o BUILD\K64F\ARM\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.o .\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.S
[DEBUG] Return: 8
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '--target=arm-arm-none-eabi'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--thumb'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-no-armcc-pragma-push-pop'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-no-armcc-pragma-anon-unions'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fdata-sections'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fno-exceptions'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--md'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fshort-enums'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-fshort-wchar'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--cpu'.
[DEBUG] Output: Fatal error: C3917U: Concatenated options cannot have arguments: -cpu cortex-m4
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '--fpu'.
[DEBUG] Output: Fatal error: C3917U: Concatenated options cannot have arguments: -fpu fpv4-sp-d16
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-M'.
[DEBUG] Output: Fatal error: C3900U: Unrecognized option '-float-abi=hard'.
[DEBUG] Output: Warning: C3910W: Old syntax, please use '-I'.
[DEBUG] Output: Fatal error: A1905U: Pre-processor step failed for '.\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.S'
[DEBUG] Output: 1 Error, 0 Warnings
Traceback (most recent call last):
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\make.py", line 74, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\build_api.py", line 486, in build_project
    objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\toolchains\mbed_toolchain.py", line 459, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\toolchains\mbed_toolchain.py", line 532, in compile_queue
    raise ToolException(err)
ToolException: Fatal error: C3900U: Unrecognized option '--target=arm-arm-none-eabi'.
Warning: C3910W: Old syntax, please use '-M'.
Warning: C3910W: Old syntax, please use '--thumb'.
Fatal error: C3900U: Unrecognized option '-no-armcc-pragma-push-pop'.
Fatal error: C3900U: Unrecognized option '-no-armcc-pragma-anon-unions'.
Fatal error: C3900U: Unrecognized option '-fdata-sections'.
Fatal error: C3900U: Unrecognized option '-fno-exceptions'.
Warning: C3910W: Old syntax, please use '--md'.
[... cut ...]
Warning: C3910W: Old syntax, please use '-I'.
Fatal error: A1905U: Pre-processor step failed for '.\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM\except.S'
1 Error, 0 Warnings

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\myTest\oob\Nucleo_blink_led"
       Command: "c:\python27\python.exe -u C:\myTest\oob\Nucleo_blink_led\mbed-os\tools\make.py -t ARM -m K64F --source . --build .\BUILD\K64F\ARM -c -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

cmonr, 2019-03-19:
> cannot find 5.12.0-rc3, it seems not be published yet.

Correct, RC3 is not available yet.

soramame21, 2019-03-19:
@ciarmcom Is it planned to fix above "Unrecognized option" at RC3? should I file another issue for it?

cmonr, 2019-03-19:
Sorry about that. Feel free to make another issue, but I think this might have something to do with the paths you have set (ARM vs ARMC6)

soramame21, 2019-03-19:
@cmonr I found following issue for the same error. please close this one. thanks.
https://github.com/ARMmbed/mbed-os/issues/10069

closed 2019-03-19 by cmonr