ARMmbed/mbed-os #10135: [OOB 5.12.0-rc2 CoreOS]Blinky with bare metal failed compiling at online-IDE

soramame21, 2019-03-18:
### Description
Blinky with bare metal failed compiling at online-IDE.
Blinky without bare metal (=remove following mbed_app.json) can be compiled without error.
### Enabling bare metal
mbed_app.json
```
{
    "requires": ["bare-metal"]
}
```
### Steps
```
1. go to https://os.mbed.com/teams/ST/code/Nucleo_blink_led/
2. click Import into Compiler bottun.
3. delete mbed
4. added library with URL, which is tag mbed-os-5.12.0-rc2
   https://github.com/ARMmbed/mbed-os/#c0939781ea1fa50657e055867193b62b1c9a6035
5. click revision button, but cannot see tag mbed-os-5.12.0-rc2 on the list?! 
6. add a new file, edit and save as above mbed_app.json
7. click Compile. got failed with following message.
Error: In /tmp/chroots/ch-edd5e9ab-38ee-457b-8743-73b4da893bfa/src/mbed_app.json element : Additional properties are not allowed ('requires' was unexpected)
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

studavekar, 2019-03-18:
Hi @soramame21,
https://os.mbed.com/teams/ST/code/Nucleo_blink_led/  is a mbed2 project please use mbed5 project.

Link for migration from mbed2 to mbed5 project https://os.mbed.com/docs/mbed-os/v5.11/tutorials/migrating-to-mbed-os-5.html

soramame21, 2019-03-19:
@studavekar, it's exactly what I'm testing, migration to mbed os 5 from online IDE following the web doc. 
Could you verify what I'm wrong?

cmonr, 2019-03-19:
> 5. click revision button, but cannot see tag mbed-os-5.12.0-rc2 on the list?! 

@soramame21 If you weren't able to select the revision, the what did you use when you attempted to compile?

soramame21, 2019-03-19:
@ciarmcom thanks for such quick reply. Here is the revision list after adding mbed library.
[image]

cmonr, 2019-03-19:
> @ciarmcom thanks for such quick reply. 

🤣 I promise I'm not one of the bots.

So that list would indicate that 5.11.5 is being used instead of 5.12.

There should be instructions on how to use the internal version of the online compiler, correct @ashok-rao ?

linlingao, 2019-03-19:
@cmonr The revision history only shows official releases. When the version used matches one of the official releases, the text becomes bold. In this case, the mbed-os version used in the program does not match any of the official releases. It would have been clearer if Ren clicked anywhere below the top line when he captured the screenshot. (I know, can this be any more confusing!)

adbridge, 2019-03-19:
@soramame21 Are you using the correct, ie oob version of the online compiler, rather than the production release ?

theotherjimmy, 2019-03-19:
@soramame21 Your screenshot shows the public online compiler. OOB uses the internal testing server use that please.

soramame21, 2019-03-20:
@theotherjimmy @adbridge @linlingao Thanks for your comments. I could not login the oob online compiler. I reset the password but no email is received although a message shown below.
`
An email has been sent to your registered email address
`

closed 2019-03-28 by ciarmcom

kamilmeric, 2019-04-03:
Hello,

I am also using mbed studio. I get same error:

[ERROR] in .\mbed_app.json element : Additional properties are not allowed ('requires' was unexpected)

i updated my os5 lib to 5.12.0

What is the problem?

cmonr, 2019-04-08:
@kamilmeric Please open a new issue with steps to reproduce.

While similar, this issue was specific to OOB testing.