ARMmbed/mbed-os #10153: LPC55S69 IAR build has conflicting RAM/ROM defines

mikisch81, 2019-03-19:
### Description

Commit cde7d0be589875fcecfc1886aa19979e8974721c added an entry for *LPC55S69JBD100* in `tools/arm_pack_manager/index.json`.

Now when building with the IAR toolchain, for example `mbed test -t IAR -m LPC55S69_NS -n mbed-os-tests-psa-spm_smoke --compile`, `MBED_ROM_START`/`MBED_RAM_START` and `MBED_ROM_SIZE`/`MBED_RAM_SIZE` are defined twice with conflicting values.

Content of `./BUILD/tests/LPC55S69_NS/IAR/./mbed-os/TESTS/psa/spm_smoke/.link_options.txt`:
```sh
--skip_dynamic_initialization --threaded_lib --config_def DOMAIN_NS=0x1
--config_def MBED_ROM_START=0x0 --config_def MBED_ROM_SIZE=0x98000 
--config_def MBED_RAM_START=0x20000000 --config_def MBED_RAM_SIZE=0x44000 
--config_def MBED_RAM_START=0x20022000 --config_def MBED_ROM_SIZE=0x68000 --config_def MBED_RAM_SIZE=0x22000 --config_def MBED_ROM_START=0x00030000
...
```

So for example `MBED_ROM_START` is defined both to `0x0` and `0x00030000` which messes the memory map and in v8m targets with security attribution units this results in Secure-faults at run-time.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

@alzix @orenc17 @theotherjimmy @bridadan @mmahadevan108

0xc0170, 2019-03-19:
This PR added the files #10068. It's marked as 5.12rc3. This is a regression in rc3 candidate.

0xc0170, 2019-03-19:
@mikisch81 Looking at the index.json changes, cant figure out where these double values come from. We might need help from above referenced people.

PR 10068 was included to rc3 candidate.  Would it be reasonable to postpone this PR to rc4 ? how urgent was to have exporters support for this LPC target ? 
How can we fix it now?

mikisch81, 2019-03-19:
@0xc0170 of course we need help from @ARMmbed/mbed-os-tools, I already sent an e-mail regarding this issue yesterday.

Regarding postponing this PR tp rc4, I think bootloader for LPC uses the index.json as well, @davidsaada can elaborate on this.

The main problem is that we do not have the CI for this target ready yet and this target is in master for quite some time.

0xc0170, 2019-03-19:
Correct, bootloader also affected. I'll try to look at index.json values and see if I can find out where double values come from.

davidsaada, 2019-03-19:
@mikisch81 I've just issued your build command on the latest mbed-os master without any issues. Are you up to date?
In addition, values in index.json file aren't used any more for building bootloader, this was introduce by  @ARMmbed/mbed-os-tools in #10113 (which was merged in).

mikisch81, 2019-03-19:
@davidsaada you were able to run the tests? Can you share the non-secure map file?

0xc0170, 2019-03-19:
I was able to reproduce this, on latest master (sha 54602f55ba). 

```
--skip_dynamic_initialization --threaded_lib --config_def DOMAIN_NS=0x1 --config_def MBED_ROM_START=0x30000 --config_def MBED_ROM_SIZE=0x68000 --config_def MBED_RAM_START=0x20000000 --config_def MBED_RAM_SIZE=0x44000 --config_def MBED_BOOT_STACK_SIZE=1024 --config_def MBED_RAM_START=0x20022000 --config_def MBED_ROM_SIZE=0x68000 --config_def MBED_RAM_SIZE=0x22000 --config_def MBED_ROM_START=0x00030000 
```

mikisch81, 2019-03-19:
@0xc0170 of course, it was never a build issue, but run-time, a secure-fault is triggered.

davidsaada, 2019-03-19:
@mikisch81 Sorry - right now can't run anything on my board (even other tests).
Was going to post map file till I saw your comment about this not being a build issue but a runtime issue.

0xc0170, 2019-03-19:
I know where these come from but dont know which values are valid.

Target overrides in target.json file

```
        "overrides": {
            "non-secure-rom-start": "0x00030000",
            "non-secure-rom-size": "0x68000",
            "non-secure-ram-start": "0x20022000",
            "non-secure-ram-size": "0x22000",
            "secure-rom-start": "0x10000000",
            "secure-rom-size": "0x28000",
            "secure-ram-start": "0x30000000",
            "secure-ram-size": "0x22000"
        },
```

And in index.json:
```
            "SRAM": {
                "access": {
                    "execute": false,
                    "non_secure": false,
                    "non_secure_callable": false,
                    "peripheral": false,
                    "read": true,
                    "secure": false,
                    "write": true
                },
                "default": true,
                "size": 278528,
                "start": 536870912,
```

This two conflicts. SRAM is set to start at 536870912 in index.json file but in "non-secure-ram-start": "0x20022000" (different values).

mikisch81, 2019-03-19:
Again this happens **ONLY** in IAR toolchain, ARM and GCC_ARM work fine.

davidsaada, 2019-03-19:
@yossi2le

yossi2le, 2019-03-19:
Sorry, but I have nothing to contribute here. 
I get the same numbers as @0xc0170  got.
I think this may be a tool issue.

theotherjimmy, 2019-03-19:
@mikisch81 @yossi2le Talk about `index.json` being correct or not is a red herring. It's about how it's used.

theotherjimmy, 2019-03-19:
The fact that it only happens on IAR is very strange...

davidsaada, 2019-03-19:
> 
> 
> The fact that it only happens on IAR is very strange...

Maybe totally off topic, but @offirko has been working on a very strange bug seen on TDBStore with SPIF on the DISCO_F746NG board, also seen on IAR8 only (not even with IAR7).

theotherjimmy, 2019-03-19:
@davidsaada That sounds like it's not something tools related (as it would probably be present on both IAR7 and IAR8 if it were)

theotherjimmy, 2019-03-19:
@mikisch81 I can only reproduce the duplication of `MBED_ROM_{START|SIZE}` I cannot reproduce the different values.

theotherjimmy, 2019-03-19:
@mikisch81 I have just worked on the command line invocation for this issue, as that's what it describes.

theotherjimmy, 2019-03-19:
That being said, I think you should give my PR a shot, as it might fix the issue.

mikisch81, 2019-03-20:
> 
> 
> That being said, I think you should give my PR a shot, as it might fix the issue.

I verified that it does 😄

PR #10157 (2019-03-20): Deduplicate IAR MBED_ROM__xxxx defines
### Description

PR #10113 allowed a porter or application author to configure 
secure or non-secure rom through the use of:
  * `target.secure-rom-start`
  * `target.secure-rom-size`
  * `target.non-secure-rom-start`
  * `target.non-secure-rom-size`

and use these in managed bootloader modes and report them as ROM. 
This created a duplicate `MBED_ROM_START` and `MBED_ROM_SIZE`
define on the IAR linker command line. 

Further, the overrides:
  * `target.secure-ram-start`
  * `target.secure-ram-size`
  * `target.non-secure-ram-start`
  * `target.non-secure [... cut]
closed 2019-03-20 by 0xc0170