ARMmbed/mbed-os #10163: QSPIFBlockDevice: SFDP bus mode parser uses wrong mask values

VairaRamasamy, 2019-03-19:
### Description

_sfdp_detect_best_bus_read_mode() parses the basic param table to find out the bus mode. It uses wrong mask values while parsing DWORD-1, byte-2. 

e.g. Bit-7 of byte-2 is 'unused'. But mask value 0x80 is used to find out if 1-1-4 is supported or not. 
 if (examined_byte & 0x80) {
            //  Fast Read 1-1-4 Supported

Bus mode | Mask used | Correct Mask
------------------------------------------
1-4-4 | 0x40 | 0x20
1-1-4 | 0x80 | 0x40
1-2-2 | 0x20 | 0x10

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

offirko, 2019-03-20:
@VairaRamasamy - my apologies. I'll fix

offirko, 2019-03-20:
Handled by #10171

closed 2019-04-01 by 0xc0170