ARMmbed/mbed-os #10165: QSPIFBlockDevice: status register write size cannot have a fixed value

VairaRamasamy, 2019-03-19:
### Description

In _sfdp_set_quad_enabled(), sr_write_size is fixed to QSPI_MAX_STATUS_REGISTER_SIZE (0x03). I found this causing issue while porting QSPI for CY8CPROTO_062_4343W to work with the NOR flash S25FL512S. For this flash chip, the write size for the status register must be 2. Otherwise, enabling quad mode fails. 

Reference: [S25FL512S datasheet](https://www.cypress.com/file/177971/download), Section: 9.3.7 Write Registers (WRR 01h). The following line makes it clear that write size must be 2. 

> CS# must be driven to the logic high state after the eighth or sixteenth bit of data has been latched. If not, the Write Registers (WRR)
> command is not executed. 

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

offirko, 2019-03-20:
sr_write_size does **not** have a fixed size. It is initialized to QSPI_MAX_STATUS_REGISTER_SIZE but can change according to QER bit. 

Actually, QSPI_MAX_STATUS_REGISTER_SIZE  was 2 and I changed it to 3 in the commit below..because turning fast_mode on other boards required writing status_reg+config_reg together (3 bytes total).

https://github.com/ARMmbed/mbed-os/commit/9b3bca89fd8295147155db035e366abd8b9c3c01#diff-8e2c0034f3ea6c90e7ecf6baef0fe7c5

 
I'll fix the code to separate status register size between 'set_quad_enable' and 'fast_mode_enable'

offirko, 2019-03-20:
Handled by: https://github.com/ARMmbed/mbed-os/pull/10171

offirko, 2019-03-20:
handled by https://github.com/ARMmbed/mbed-os/pull/10171

VairaRamasamy, 2019-03-26:
@offirko is there a plan to support 4-bytes addressing anytime soon?

offirko, 2019-03-26:
@VairaRamasamy : we have considered adding support for 4-bytes addressing once we see demand for SPI Flash devices with larger capacity.
Do you require 4-bytes addressing for flash larger than 16MB?

VairaRamasamy, 2019-03-26:
Yes. We need it. Cypress' flash chips used on PSoC 6 kits support up to 64 MB.

offirko, 2019-03-26:
@VairaRamasamy - thanks for the information. We can try to make into 5.13 
I've added in CC the relevant people for planning.

CC: @dannybenor , @ARMmbed/mbed-os-maintainers

closed 2019-04-01 by 0xc0170