ARMmbed/mbed-os #10167: UBLOX_N2XX driver missing provide-default

juhoeskeli, 2019-03-20:
### Description

The UBLOX_N2XX driver is missing provide-default / get_default_instance() mechanism.

Fix example:
In UBLOX_N2XX.cpp:
```
#if MBED_CONF_UBLOX_N2XX_PROVIDE_DEFAULT
#include "UARTSerial.h"
CellularDevice *CellularDevice::get_default_instance()
{
    static UARTSerial serial(MBED_CONF_UBLOX_N2XX_TX, MBED_CONF_UBLOX_N2XX_RX, MBED_CONF_UBLOX_N2XX_BAUDRATE);
#if defined (MBED_CONF_UBLOX_N2XX_RTS) && defined(MBED_CONF_UBLOX_N2XX_CTS)
    tr_debug("UBLOX_N2XX flow control: RTS %d CTS %d", MBED_CONF_UBLOX_N2XX_RTS, MBED_CONF_UBLOX_N2XX_CTS);
    serial.set_flow_control(SerialBase::RTSCTS, MBED_CONF_UBLOX_N2XX_RTS, MBED_CONF_UBLOX_N2XX_CTS);
#endif
    static UBLOX_N2XX device(&serial);
    return &device;
}
#endif
```

FYI:
@mudassar-ublox 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

mudassar-ublox, 2019-03-21:
Pull request is created. 
https://github.com/ARMmbed/mbed-os/pull/10179

0xc0170, 2019-03-22:
Integrated, closing

closed 2019-03-22 by 0xc0170