ARMmbed/mbed-os #10168: Bin file over 2GB size if specifying QSPI region

janjongboom, 2019-03-20:
### Description

Some STM32 boards can program both the internal flash and on-board QSPI from a single binary. You do this by specifying a new linker region at address `0x90000000` and placing parts of the code in here. However, when building this binary with Mbed OS you'll end up with a 2GB image. Can we somehow avoid this? Maybe by using .hex format? If so, how could I specify this?

This is on Mbed OS 5.11.5, e.g. on stm32f746-disco.

Issue was originally found by Etteplan.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-03-22:
> Can we somehow avoid this? Maybe by using .hex format? If so, how could I specify this?

Use `"OUTPUT_EXT": "hex",` in targets.json file. 

How would the linker sectors look like? Assuming there's that big gap between internal address space and QSPI, it's being filled in the binary so hex should help in that case  - have you tested it?

theotherjimmy, 2019-03-26:
This is not an issue with master, or any release of mbed-os, as you're using your own linker script. @0xc0170's advice may also apply to the `target_overrides` section of your `mbed_app.json`. This should allow you to work around this in your project @janjongboom.

closed 2019-05-22 by ciarmcom

0xc0170, 2019-05-22:
> This is not an issue with master, or any release of mbed-os, as you're using your own linker script. @0xc0170's advice may also apply to the target_overrides section of your mbed_app.json. This should allow you to work around this in your project @janjongboom.

Closed as "cannot reproduce"