ARMmbed/mbed-os #10169: Mainline Mbed HTTP

janjongboom, 2019-03-20:
### Description

Mbed OS is an operating system for IoT devices, and a lot of IoT devices connect to the internet through HTTP / HTTPS. But Mbed OS does not contain an HTTP implementation, whilst there is a CoAP one (and soon an MQTT one I believe). About a year and half ago I wrote [mbed-http](https://os.mbed.com/teams/sandbox/code/mbed-http/) which is now widely used, including in partner projects ([1](https://os.mbed.com/users/atomichan/code/Wio_3G_HTTP-POST-example/), [2](https://os.mbed.com/teams/Renesas/code/GR-PEACH_IoT_Platform_HTTP_sample/), [3](https://os.mbed.com/teams/Avnet/code/http-example-wnc/), [4](https://os.mbed.com/teams/Cloud-Hackathon/code/IFTTT-Google-Sheets-Demo/)), in the Treasure Data documentation ([5](https://support.treasuredata.com/hc/en-us/articles/360012567313)), and soon also in our own docs ([6](https://github.com/ARMmbed/mbed-os-examples-docs_only/pull/17)).

I think that this project should move into Mbed OS core. It would mean that it could be part of CLI, is guaranteed to be updated when the OS updates, and it can serve as a set of real-world tests around the network interface APIs for many boards. It already uses TLSSockets underneath, and it has Greentea tests.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

cmonr, 2019-03-20:
@SenRamakri

cmonr, 2019-03-20:
@ARMmbed/mbed-os-ipcore @ARMmbed/mbed-os-core

SeppoTakalo, 2019-03-21:
Can this be created as a PREQ for Mbed OS project?

janjongboom, 2019-04-24:
@cmonr @bulislaw Why is this closed?

40Grit, 2019-04-24:
@0xc0170 Still a lot of issues closed that should be open. Including the ones I mentioned last week.

@linlingao

closed 2019-10-31 by ciarmcom