ARMmbed/mbed-os #10176: waffle closing

bmcdonnell-ionx, 2019-03-20:
### Description

README.md contains info from and links to [Waffle, which is shutting down on May 16, 2019](https://waffle.io/closing-its-doors). I suppose you'll want to remove those links/info by that time, and potentially migrate whatever info/services it's providing for you elsewhere.

Do you have a plan in place?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-03-20:
Ah, you have a good point.

I don't think we have a good way to generate an SVG live, so I'm thinking it would be simplest to remove the two lines.

@ARMmbed/mbed-os-maintainers Thoughts?

0xc0170, 2019-03-21:
Hi, I've got an email in the last days about it. We will remove it.

closed 2019-03-27 by cmonr

sg-, 2019-03-28:
Looks like not everything was removed in https://github.com/ARMmbed/mbed-os/commit/88c33a09c0bc438c6bbf9e47766d007ee4309ed0

[image]

cmonr, 2019-03-28:
@sg- Thanks for the find!

PR opened: https://github.com/ARMmbed/mbed-os/pull/10260

PR #10194 (2019-03-27): readme: remove waffle as not supported soon
### Description

Fixes #10176

We will need to find alternative, will reopen 10176 to keep track of it. 

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
commit 88c33a09 (2019-03-27): readme: remove waffle as not supported soon

Fixes #10176