ARMmbed/mbed-os #10183: mbed_ram_start/size is not sent to CY8CKIT_062_WIFI_BT_PSA

orenc17, 2019-03-21:
### Description

when compiling anything for CY8CKIT_062_WIFI_BT_PSA the MBED_RAM_START/SIZE is missing from both CC and LD.

this issue causes the board to fail directly on boot
@theotherjimmy 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-03-21:
Fix needed in 5.12rc4?

orenc17, 2019-03-21:
yes the board does not boot

orenc17, 2019-03-21:
investigated a bit,
the issue is that CY8CKIT_062_WIFI_BT does not specify RAM regions in cmsis-pack-manager
resulting in get_all_active_memories() returning an empty dict

@theotherjimmy i think PSA ram & rom regions should be handled prior to _get_primary_memory_override()

plus we will need to find a solution for platforms that are not in cmsis-pack-manager

theotherjimmy, 2019-03-21:
@orenc17 `_get_primary_memory_override()` handles "PSA" ram and rom, and regular ram and rom. We should call it earlier maybe.

theotherjimmy, 2019-03-21:
@orenc17 Have you tried a target without a CMSIS Pack? I'm pretty sure that if it has any of the other start/sizes defined it'll work.

theotherjimmy, 2019-03-21:
@orenc17 This is not so simple: IROM1 is all of:
* readable
* writable
* executable

Making it match both the filter for ROM and RAM. This does some weird things with the logic in `get_all_active_memories`

PR #10186 (2019-03-21): Remove cypress boards from CMSIS pack manager
### Description

Remove cypress boards from CMSIS pack manager as the values were fabricated and are no longer necessary. (@dannybenor can confirm that)
 
This fixes #10183 

depending on:
* #10187  
* #10188 

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers
@theotherjimmy @ARMmbed/team-cypress 

### Release Notes
PR #10187 (2019-03-21): Fall back to configuration information when CPM reports no memories
### Description

The CY8CKIT targets have a very special memory layout that includes a 
memory that is all of:
 * Read
 * Write
 * Execute

This makes CPM report this memory as both a RAM and a ROM. However,
When reported as a RAM, this memory will be filtered out by the build
tools, leaving this target without a defined RAM.

This Pull request corrects the logic to look for a RAM elsewhere in
configuration when this happens.

Fixes #10183

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Funct [... cut]
closed 2019-03-21 by 0xc0170