ARMmbed/mbed-os #10195: BLE: SM can't get control block in onConnectionComplete event

desmond-blue, 2019-03-22:
### Description

When BLE connection complete, `_eventHandler->onConnectionComplete` gets called before `processConnectionEvent`, thus `acquire_control_block` gets called before `get_control_block`, that makes `_ble.securityManager().setLinkSecurity()` failed.

```
 PERIPHERAL

da:52:cc:42:72:e5
Please connect to device
Connected to peer: 73:71:47:83:00:34
Error during SM::setLinkSecurity - : BLE_ERROR_INVALID_PARAM: One of the supplied parameters is invalid
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

simonpfeifhofer, 2019-10-08:
I've the same problem.
@ciarmcom any updates on this?

talorion, 2019-10-08:
I've the same problem.
any updates on this?

pan-, 2020-06-01:
@desmond-blue From my understanding, this has been fixed, can you confirm ?

adbridge, 2021-06-07:
> @desmond-blue From my understanding, this has been fixed, can you confirm ?

No response to this so I am closing accordingly.

closed 2021-06-07 by adbridge