ARMmbed/mbed-os #10197: Slow context switch after releasing a semaphore

linlingao, 2019-03-22:
### Description
Cypress reports:
Basically after releasing a semaphore, the scheduler seems to just wait for something between 6-12ms before doing anything. This is completely crippling for WIFI throughput.
Here are the details on the semaphore/Scheduler issue with MbedOS.

1. WDD thread takes a semaphore and blocks. The semaphore wait has a 4ms timeout. This thread is running at the highest thread priority in the system.

2. Upon receiving interrupt, the ISR gives the semaphore.

3. This should unblock and schedule the WDD thread to run.

4. But WDD thread takes between 6 – 12 ms to run after semaphore is given by the ISR for almost every 3 packets

Target: pSoC6

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-03-24:
cc @JonatanAntoni

kjbracey-arm, 2019-03-25:
Whatever the issue is here, it's not generally happening, as I know we have drivers such as the Atmel RF233 which are routinely managing to do `InterruptIn` -> `EventFlag` -> High priority thread -> SPI transaction within a 192us hard realtime limit to receive a radio ACK.

So I'd like more info about the timing. Are you sure the 6-12ms is measured from the ISR actually releasing the semaphore? Or is it measured from the interrupt input?

The Cypress `hal_sleep` function appears to be a black box. Are you sure it's waking immediately, and that the 6ms-12ms isn't a wake delay from that sleep? Could you try replacing `hal_sleep` with an empty function, and then with a plain `__WFI`? Could you maybe be deep sleeping when you shouldn't be? Deep sleep lock not held?

You say the high-priority thread has a 4ms timeout - what is it doing after the timeout? Are you sure the thread is actually waiting for the semaphore for the 6ms-12ms, not busy processing other work?

> for almost every 3 packets

Does this mean on every third packet? So two out of three wake in time?

linlingao, 2019-03-25:
@satya1957 Can you answer questions from @kjbracey-arm ?

linlingao, 2019-03-25:
CC @ARMmbed/team-cypress

balajicyp, 2019-03-25:
Hi Kevin,

My Answers are inline.. 

So I'd like more info about the timing. Are you sure the 6-12ms is measured from the ISR actually releasing the semaphore? Or is it measured from the interrupt input?

>> Yes the ISR happens immediately which signals the waiting thread on that semaphore, but the semaphore wakes up the thread after 6 or 11ms  every 3rd of the packet...

The Cypress hal_sleep function appears to be a black box. Are you sure it's waking immediately, and that the 6ms-12ms isn't a wake delay from that sleep? Could you try replacing hal_sleep with an empty function, and then with a plain __WFI? Could you maybe be deep sleeping when you shouldn't be? Deep sleep lock not held?
>> we haven't tried replacing hal_sleep with __WFI. There is no deep sleep functionality in the code being tested..

You say the high-priority thread has a 4ms timeout - what is it doing after the timeout? Are you sure the thread is actually waiting for the semaphore for the 6ms-12ms, not busy processing other work?

>>  The higher priority thread waits for max 4ms timeout or reads a SDIO packet within 4ms and passes it on to network stack. This is dedicated thread listening for packets from SDIO bus, it is not doing anything else.. What I see from the entry and exit times of bus read packet is that it takes 0 ms for (3 consecutive 1586 bytes of packet) and there is a sudden jump of 6 to 13 ms after that. I have code which takes average of the last 100 packets which comes out to 6 ms delay.

Thanks
Balaji.

kjbracey-arm, 2019-03-26:
Which target is this, precisely? Is `MBED_TICKLESS` on?  If `MBED_TICKLESS` is on, then deep sleep could be being entered, and maybe that is the problem, if you're not holding a deep sleep lock.

> Yes the ISR happens immediately which signals the waiting thread on that semaphore, but the semaphore wakes up the thread after 6 or 11ms every 3rd of the packet...

And you're instrumenting this how - wiggling a `DigitalOut` in the ISR handler or something?

I would recommend doing this:

    // ISR handler
    // DigitalOut isr_pin(D5);
    isr_pin = 1;
    semaphore.release();
    isr_pin = 0;

    // Thread
    // DigitalOut thread_pin(D6);
    thread_pin = 1;
    semaphore.wait();
    thread_pin = 0;

Making sure to bracket each call with a transition - to confirm that the thread was waiting, not doing something else.

> I have code which takes average of the last 100 packets which comes out to 6 ms delay.

So *every* packet after the third takes 6ms?

I think you need to break this down a bit. On this platform, if you just have a ticker for the ISR handler, do you see a problem with it releasing a semaphore and handing over to a thread? Or is the problem only visible in the full wifi driver, which would suggest something there is interfering.

0xc0170, 2019-05-06:
> I think you need to break this down a bit. On this platform, if you just have a ticker for the ISR handler, do you see a problem with it releasing a semaphore and handing over to a thread? Or is the problem only visible in the full wifi driver, which would suggest something there is interfering.

Very good point. @balajicyp Do you have a small example to reproduce that would indicate rtos issue or is this application specific?

maclobdell, 2019-06-19:
@linlingao  Do you have a testcase that can show this issue?  

There is a new pull request https://github.com/ARMmbed/mbed-os/pull/10692 that updates the Cypress WiFi to a new implementation called WHD.  

It would be great to run a test case on the WDD verses the new WHD and confirm if this issue is fixed.

linlingao, 2019-06-19:
@maclobdell I don't have a test case. @0xc017 asked @balajicyp for one but we haven't heard back.

maclobdell, 2019-08-15:
@linlingao @balajicyp - is this resolved with the latest release based on the new hal implementation?  Can this be closed?

0xc0170, 2019-08-21:
I'll close this issue as we don't have details how to reproduce. If we get more details, we can reopen.

closed 2019-08-21 by 0xc0170

balajicyp, 2019-10-31:
Yes, Please close this issue.