ARMmbed/mbed-os #10203: TEMPSENSOR_CAL1_TEMP constant wrong value in stm32f3xx_ll_adc.h

ecoromka, 2019-03-23:
### Description

It seems to be a a bug in defenition of TEMPSENSOR_CAL1_TEMP constant. https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.11.5/targets/TARGET_STM/TARGET_STM32F3/device/stm32f3xx_ll_adc.h#L389
Datasheet says it is 30 while in file above it's 25. It leads to an error in internal temperature readings, using provided macro.
Datasheet, page 150/173, Table 90: https://www.st.com/resource/en/datasheet/stm32f303re.pdf

I've checked datasheets for all MCU, mentioned in file above (STM32F301x8 || STM32F302x8 || STM32F302xC || STM32F302xE || STM32F303x8 || STM32F303xC || STM32F303xE || STM32F318xx || STM32F328xx || STM32F334x8 || STM32F358xx || STM32F398xx), and none of those says any different: only 30 Celcius for TS_CAL1 value.

For some reason, in other block, at line 7475, the value is 30, as expected.

Don't know why GitHub doesn't show the file in search reults. Other targets seems to be OK.

Regards, Roman

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-03-25:
Hi @ecoromka 
I confirm the issue.
If you push a pull request , I will approve it.
Thx for report

jeromecoutant, 2019-03-25:
@0xc0170 you can set device ST label

ST_INTERNAL_REF 63882

jeromecoutant, 2019-04-08:
Hi
Please close this issue as #10212 is merged
Thx

ecoromka, 2019-04-08:
Thanks for your help!

closed 2019-04-08 by ecoromka