ARMmbed/mbed-os #10207: The can.write called from Ticker or EventQueue cause a crash

JohnK1987, 2019-03-24:
### Description
Hi,
I found an issue when I try to use this code on my ST board with latest MbedOS 5
https://os.mbed.com/teams/mbed_example/code/CAN_ex_1/file/d6cbcdb04ada/main.cpp/ 
This code crash with this error
/////////////////////////////////////////////////////////////////////////////////////////////////
++ MbedOS Error Info ++
Error Status: 0x80020115 Code: 277 Module: 2
Error Message: Mutex lock failed
Location: 0x8007DEB
Error Value: 0xFFFFFFFA
Current Thread: rtx_idle  Id: 0x200005AC Entry: 0x8008AE5 StackSize: 0x300 StackMem: 0x20000638 SP: 0x2007FF18 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020115
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=2) reached maximum, system will halt after rebooting [00]
////////////////////////////////////////////////////////////////////////////////////////////////

It looks like the can.write(CANmsg) method cause the issue but only when it is called from Ticker or EventQueue(tested in my personal project), because when the can.write(CANmsg) method is called from the main, it works well.
This issue not occur with latest MbedOS 2 library version but occur with higher version of the MbedOS 5 then "origin/mbed-os-5.9 tag:mbed-os-5.9.7" from  10. sep 2018. This is the last functional version.

Best regards
J.

Tools and equipment:
- Nucleo-F767ZI and F446RE
- home made shield with 2 x can-bus transceiver mcp2551 and necessary components
- Mbed Compiler Online 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

kjbracey-arm, 2019-03-25:
This is a generic rule for the C++ driver classes - they enforce ordered use of the HAL with a mutex, and because of that mutex, they can't normally be used from interrupt context. You should normally make the interrupt attachment defer work to a thread. That's generally a good idea, as it should help interrupt latency - if the operation you're doing is slow, you might end up losing serial data by blocking UART interrupts, for example.

Previous versions of mbed OS let the mutex lock silently fail from interrupt context - somewhere around 5.9 it started being trapped in debug builds, and it's now trapped in develop builds too.

You can make your own thread, or use a global event queue. Something like

    can.attach(mbed_highprio_event_queue()->event(my_callback));

However, I recall reading a issue that certain HALs wouldn't permit this because returning from the interrupt handler cleared received data along with the interrupt condition, so `read` would not work called later from a thread. I can't now find that issue, but I would say that was a fault in those HALs.

Anyway, if you do really want to your operation from interrupt, you can do it by making an `UnlockedCAN` object. This pattern works for most HAL drivers: derive from the class and override `lock` and `unlock` with empty methods. Then serialization of calls to that HAL object becomes your responsibility. (You can see an example of `UnlockedSPI` in `NanostackRfPhyAtmel.cpp`.)

jeromecoutant, 2019-04-12:
@JohnK1987 - did it answer your issue ?
Thx

JohnK1987, 2019-04-15:
Hi, thx for your responce and sorry for delay.
> Anyway, if you do really want to your operation from interrupt...

I tried the UnlockedCAN how kjbracey-arm wrote
```c++
class UnCan : public CAN {
public:
    UnCan(PinName rd, PinName td) : CAN(rd, td) { }
    virtual void lock() { }
    virtual void unlock() { }
};
```
That was working until new version(5.12) of MbedOS5 came. So the easier way for me is use old version(5.9) where was implemented how kjbracey-arm wrote than again and again try to solve new changes.

I reworked original example: 
[CAN-BUS loop back with Ticker flag](https://os.mbed.com/users/JohnnyK/code/CANLoop_ticker/file/c1f6e9d285c7/main.cpp/)
[CAN-BUS loop back with EventQueue](https://os.mbed.com/users/JohnnyK/code/CANLoop_EventQueue/file/d31fd55babb2/main.cpp/)
These both work well, only when I try to replace  polling variant "can2.read(msg)", it always freeze.

> This is a generic rule for the C++ driver classes...

So if I understand correctly the code above (in my original post) is wrong. So why is this code used as example in the documentation for CAN as [CAN hello, world](https://os.mbed.com/docs/mbed-os/v5.12/apis/can.html)? This is very confusing, especially for novices or beginners.

I'm not a professional, so I apologize if I don't understand it correctly.

Best regards
Jan

bulislaw, 2019-05-07:
@JohnK1987 I'll close the issue for now, please reopen it if you have further questions.

closed 2019-05-07 by bulislaw