ARMmbed/mbed-os #10216: Support osWaitForever in Mail::alloc() and Mail::calloc()

desowin, 2019-03-25:
### Description

The Mail class provides two functions to allocate message buffer: `alloc()` and `calloc()`. Both of these functions take optional `millisec` parameters which defaults to 0. The API itself seems OK, expect for the fact that millisec parameter is not used at all (which is documented).

In other RTOSes, it is possible to block when putting data to statically allocated mailbox. For example:

- SEGGER embOS: the function `OS_PutMail()` blocks if message cannot be copied into mailbox buffer (the non-blocking option is `OS_PutMailCond()`, and timeout is possible to be specified with `OS_PutMailTimed()`)
- FreeRTOS: Statically allocated queue can be created using `xQueueCreateStatic()`. The parameter `xTicksToWait` for `xQueueSend()` can be `portMAX_DELAY`. If the `INCLUDE_vTaskSuspend` is active, then the `xQueueSend()` will block indefinitely otherwise it'll wait for the maximum possible timeout).

I think the easiest way to implement the wait in mbed `Mail::alloc()` and `Mail::calloc()` (if memory pool is exhausted) would be to include Semaphore in the Mail class. The Semaphore would be initialized to `queue_sz` and `Semaphore::wait()` would be called in `Mail::alloc()` and `Mail::calloc()`, while the `Semaphore::release()` would be called in `Mail::free()`.

The only problem with above proposal is the fact that Mail `queue_sz` is `uint32_t`, while the Semaphore's max count is `uint16_t`.

What would be the best way to implement blocking in `Mail::alloc()` and `Mail::calloc()`?

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

desowin, 2019-03-25:
Apparently, the `osSemaphoreNew()` supports 32-bit unsigned maximum count. So it would be possible to use this API directly in the `Mail` without changing the `queue_sz` type.

cmonr, 2019-03-25:
@ARMmbed/mbed-os-core

kjbracey-arm, 2019-03-26:
It seems `Semaphore`  wants to be extended to permit 32-bit maximum counts. I'll make a patch for that.

But for `Mail`, I don't see why isn't it just passing the timeout to `MemoryPool` - the underlying `osMemoryPoolAlloc` it uses takes a timeout.

I've never used any of `Mail`, `MemoryPool` or `Queue`, so I don't know what the historical reason for ignoring the timeout was.

I'll make a patch to add that to the API and see who comments.

PR #10225 (2019-04-12): Mail/MemoryPool: blocking alloc
### Description

`Mail` and `MemoryPool` did not permit access to the blocking form of allocation available in CMSIS-RTOS 2 - give them new blocking methods.

Fixes #10216.

### Pull request type

    [ ] Fix
    [ ] Refactor
    [ ] Target update
    [X] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Release Notes

* `Mail` and `MemoryPool` now provide blocking forms of `alloc` calls that can sleep the thread until a memory block is available.
closed 2019-04-12 by cmonr