ARMmbed/mbed-os #10223: TARGET_DISCO_L475VG_IOT01A unexpected serial port behaviour

jupe, 2019-03-26:
### Description

`printf()` behaviours unexpectedly for this board. Seems that it depends on line feed does it write data to serial port or not.

- What target does this relate to?
TARGET_DISCO_L475VG_IOT01A

- What toolchain (name + version) are you using?
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

- What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
   mbed-cli @ 1.8.3

- What is the SHA of Mbed OS (git log -n1 --oneline)?
https://github.com/ARMmbed/mbed-os/#6906f39bf586aaafac1d0a23eadb604c0905c476

- Steps to reproduce.

This does not print anything to serial port:
```
#include "mbed.h"
int main() {
    printf("test");
}
```

but this does (adding just one linefeed end of print):
```
#include "mbed.h"
int main() {
    printf("test\n");
}
```

    [ ] Question
    [ ] Enhancement
    [x] Bug

tommikas, 2019-03-26:
Happens also with ARM Compiler 5.06 update 6.

jupe, 2019-03-26:
looks like printf/serial works as a line buffered even it should be character buffered. But even I switch explicitly `platform.stdio-buffered-serial` to `false` it does not write anything. I also expected that if `platform.stdio-flush-at-exit` are true above application would flush data to serial port regardless buffered mode, but that does not happen. Guessing that there might be two different bugs, one related to `platform.stdio-flush-at-exit` and one related to `platform.stdio-buffered-serial`.

LMESTM, 2019-03-26:
@jupe I don't think this specifically relates to DISCO_L475VG_IOT01A - I guess this is rather mbed-os generic.  If you confirm, would you update the title to be more generic ? 

@0xc0170  any comment ?

kjbracey-arm, 2019-03-26:
The buffering you're seeing is inside the C library. stdout is allowed to be either line-buffered or non-buffered by default, if going to the console:

> As initially
> opened, the standard error stream is not fully buffered; the standard input and standard
> output streams are fully buffered if and only if the stream can be determined not to refer
> to an interactive device.

I don't know what each of our C libraries choose for line-buffering versus no buffering.

If you need to ensure there's no buffering for your app, then `setbuf(stdout, NULL)` or `setvbuf(stdout, NULL, _IONBF, 0)` will do it.

jupe, 2019-03-27:
I can confirm that same behaviour happens with K64F for example. Also tried `setbuf` which seems to work as expected. See example:
```
int main() {
    setbuf(stdout, NULL);
    setvbuf(stdout, NULL, _IONBF, 0);
    printf("payload");
}
```

This also seems to flush data to serial port as expected when `platform.stdio-flush-at-exit` is true:
```
int main() {
    printf("payload");
    exit(0);
}
```
`payload` is not printed when `platform.stdio-flush-at-exit` is `false` . This seems to be correct behaviour.
Question: should stdout flush happen even `exit()` is not called explicitly in above case ?

Question: What does `platform.stdio-buffered-serial` mean in practical? At least some buffering happens somewhere even flag is set to `false`, but where ?
[Documentation says](https://os.mbed.com/docs/mbed-os/v5.8/reference/configuration-platform.html):
> Description: Use UARTSerial driver to obtain buffered serial I/O on stdin/stdout/stderr. If false, unbuffered serial_getc and serial_putc are used directly.

Originally I did expect that when `platform.stdio-buffered-serial` is set to `false` I would receive all data from serial port even data does not contains line feeds (no any (line/char) buffering), but is this assumption wrong ? Maybe some clarification would be good for documentation (or I did not find it yet).

kjbracey-arm, 2019-03-28:
I think our documentation hasn't really considered the C library buffering, as it's not something we have much control over, as a system, and isn't Mbed OS specific. The application can control it via C library calls. I guess it should be noted somewhere.

> What does platform.stdio-buffered-serial mean in practical? At least some buffering happens somewhere even flag is set to false, but where ?

The C library has buffers for streams, and it can be configured to call the OS (`FileHandle` object) for every character, every  line, or only when its buffer is full/exhausted. That's independent of `stdio-buffered-serial`. That setting controls what happens when the C library *does* output something to the OS - is the OS's serial device buffered or not.

When it's set to off, then characters go out synchronously to the serial port, waiting for every character. Control returns to the application only when every character is in the hardware. You will stall waiting for hardware on nearly every print, unless it fits into the single-byte data register and any extra FIFO there is. (There often is no FIFO on these chips).

When it's set to on, there is a software buffer and an interrupt pump for it. When the C library outputs data, it's transferred to the software interrupt buffer, and that will be pumped out to hardware in the background later via interrupts. So you won't necessarily stall on a print if there is software buffer space available. You'll only stall if we need to wait for more buffer space.

The buffering also means you can receive data reliably - received data goes into the software buffer under interrupt so it's ready to read from `stdin` later. If the buffering is off, then arriving data is lost unless you happen to already be spinning in a read call waiting.

> should stdout flush happen even exit() is not called explicitly in above case ?

According to the C and C++ standards, returning from `main` is supposed to be the same as calling `exit`, so the behaviour should be consistent. And the C library is supposed to close (and hence flush) stdout and stderr on exit anyway:

> Next, all open streams with unwritten buffered data are flushed, all open streams are closed, and all files created by the tmpfile function are removed.

I guess we've hijacked `exit` and made that behaviour optional to reduce code size?

Looking at it, I think `main` is not being treated specially in the RTOS build. It's just another thread. If `main` exits, then the main thread terminates, but the program does not. The RTOS carries on running.  `exit` is not called when RTOS `main` returns, so there's no flush. I'm not sure what the situation is for bare-metal builds.

Changing that could be a significant compatibility issue, but I can imagine a new `rtos.exit_on_return_from_main` option.

evedon, 2020-01-10:
It does not appear that there is a Mbed OS issue so I am closing this ticket.
This is more of a useful discussion on the C library buffering.

closed 2020-01-10 by evedon