ARMmbed/mbed-os #10226: Compile error with enabling crash report

masoudr, 2019-03-26:
### Description

Hi,
I'm trying to set auto reboot for fault errors with help of [this](https://os.mbed.com/docs/mbed-os/v5.11/apis/error-handling.html) document. But when I try to compile I get the following compile error which seems generate when enabling `crash-capture-enabled` in configuration.
I tested with mbed-os `5.11.5`, `5.12.0`. My target platform is an `STM32F407VG`.
`mbed_app.json`:
```
    "target_overrides": {
        "*": {
            "platform.crash-capture-enabled": true,
            "platform.fatal-error-auto-reboot-enabled": true,
            "platform.error-reboot-max": 10,
        }
```
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o:../.\mbed-os\cmsis\TARGET_CORTEX_M/mbed_fault_handler.c:36: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o: in function `mbed_fault_handler':
> ../.\mbed-os\cmsis\TARGET_CORTEX_M/mbed_fault_handler.c:79: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o: in function `print_context_info':
>../.\mbed-os\cmsis\TARGET_CORTEX_M/mbed_fault_handler.c:142: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/platform/mbed_error.o: in function `mbed_error_initialize':
> .././mbed-os/platform/mbed_error.c:235: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/platform/mbed_error.o: in function `mbed_error':
> .././mbed-os/platform/mbed_error.c:302: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/platform/mbed_error.o:.././mbed-os/platform/mbed_error.c:348: more undefined references to `__CRASH_DATA_RAM_START__' follow
> collect2.exe: error: ld returned 1 exit status
> [ERROR] c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o../.\mbed-os\cmsis\TARGET_CORTEX_M/mbed_fault_handler.c:36: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o: in function `mbed_fault_handler':
> ../.\mbed-os\cmsis\TARGET_CORTEX_M/mbed_fault_handler.c:79: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/cmsis/TARGET_CORTEX_M/mbed_fault_handler.o: in function `print_context_info':
> ../.\mbed-os\cmsis\TARGET_CORTEX_M/mbed_fault_handler.c:142: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/platform/mbed_error.o: in function `mbed_error_initialize':
>.././mbed-os/platform/mbed_error.c:235: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/platform/mbed_error.o: in function `mbed_error':
> .././mbed-os/platform/mbed_error.c:302: undefined reference to `__CRASH_DATA_RAM_START__'
> c:/program files (x86)/gnu tools arm embedded/8 2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/DISCO_F407VG/GCC_ARM-DEBUG/mbed-os/platform/mbed_error.o:.././mbed-os/platform/mbed_error.c:348: more undefined references to `__CRASH_DATA_RAM_START__' follow
> collect2.exe: error: ld returned 1 exit status
### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2019-03-26:
@SenRamakri

SenRamakri, 2019-04-24:
@masoudr - Have you made corresponding changes to your linker scripts according to section "Adding the Crash-data-RAM region for crash reporting" in https://os.mbed.com/docs/mbed-os/v5.11/apis/error-handling.html?

masoudr, 2019-04-24:
@SenRamakri Thanks, I didn't know that. I read the docs but I couldn't realize how to apply it, I'm using GCC ARM Compiler. I found the linker file corresponding to my device: `...\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F407xG\device\TOOLCHAIN_GCC_ARM\STM32F407XG.ld` but where should I put `__CRASH_DATA_RAM_START__ ` and `__CRASH_DATA_RAM_END__ `?

masoudr, 2019-04-24:
I compiled this sample code successfully but the device gets freezes and isn't rebooted. I also don't see any information about the exception:
```
#include "mbed.h"
#include "mbed_error.h"
#include "mbed_fault_handler.h"

void generate_bus_fault_unaligned_access()
{
    SCB->CCR |= SCB_CCR_UNALIGN_TRP_Msk;
    uint32_t inv_addr = 0xAAA3;
    uint32_t val = *(uint32_t *)inv_addr;

    printf("\nval= %X", val);

    return;
}

int main()
{
    wait(2);
    generate_bus_fault_unaligned_access();
}
```

bulislaw, 2019-05-07:
@masoudr did you manage to fix that issue? I'm afraid we won't be able to investigate it anytime soon. If you have time and it's an important issue for you please consider contributing a fix.

masoudr, 2019-05-07:
@bulislaw Yes, I get it working by increasing the `rtos.main-thread-stack-size`. I will make a PR for it.

closed 2019-05-07 by masoudr

jeromecoutant, 2019-05-09:
FYI: #10365