ARMmbed/mbed-os #10227: Build failure in exported mbed-os-example-mesh-minimal in KEIL uvision5

artokin, 2019-03-26:
### Description

Exported example application https://github.com/ARMmbed/mbed-os-example-mesh-minimal fails to build using Keil uvision5.

Steps to reproduce:

1. Install Keil uvision5 that supports ARMC6 compiler >= 6.11
1.  Clone https://github.com/ARMmbed/mbed-os-example-mesh-minimal
1. Run `mbed deploy`
1. Change mbed-os revision to `master` or `mbed-os-5.12.0-rc4`
1. Export project ` mbed export -i uvision6 -m k64f`
1. In Keil uvision5 `Project` menu run `Build Target` command
1. Observe results

**Expected result:**
- Application can be build successfully

**Actual result:**
Build fails with lot of errors:
`".\BUILD\mbed-os-example-mesh-minimal.axf" - 1117 Error(s), 89 Warning(s).`

There seems to be clashing file `mbed-os/features/storage/FEATURE_STORAGE/cfstore\version` that is accidentally build in:
`
mbed-os/drivers/AnalogIn.cpp(18): warning: In file included from...
mbed-os\drivers/AnalogIn.h(25): warning: In file included from...
mbed-os\platform/SingletonPtr.h(29): warning: In file included from...
C:\Keil_v5\ARM\ARMCLANG\Bin\..\include\libcxx\new(84): warning: In file included from...
C:\Keil_v5\ARM\ARMCLANG\Bin\..\include\libcxx\exception(82): warning: In file included from...
C:\Keil_v5\ARM\ARMCLANG\Bin\..\include\libcxx\type_traits(406): warning: In file included from...
mbed-os/features/storage/FEATURE_STORAGE/cfstore\version(1): error: expected unqualified-id
{
^
`
If file `mbed-os/features/storage/FEATURE_STORAGE/cfstore\version` is cleared then build succeeds.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2019-03-26:
> Export project mbed export -i uvision6 -m k64f

@artokin Should the above quoted line have been 'uvision5' instead?

artokin, 2019-03-27:
@cmonr , shouldn't we use `uvision6` to get application build with ARMC6? At least the application can be built OK when `cfstore/VERSION` is cleared.

If parameter `uvision5` is used then the project generation fails:

```
~/github/mbed-os-example-mesh-minimal ((mbed-os-5.12.0-rc4))$ mbed export -i uvision5 -m k64f
K64F not supported by uvision5
[mbed] Auto-installing missing Python modules...
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 2
```

j3hill, 2019-03-28:
@artokin If you installed the latest uVision "mdk527.exe", this actually installs the ARM 6.12 compiler, not the ARM 6.11 compiler (...you can see this if you go to the "Help" information in the uVision IDE). This issue also only occurs on Windows build systems because the "VERSION" file referenced above has the same name as another "version" file that is created and placed in that folder (...and Windows is not case sensitive). If you rename the "VERSION" file in ``mbed-os/features/storage/FEATURE_STORAGE/cfstore`` something like "VERSION5" you should be able to compile.

artokin, 2019-04-02:
@j3hill , I would like to get Mbed OS fixed :-)
No tricks should be needed for `uvision5` nor `uvision6`

0xc0170, 2019-04-02:
Fixed via https://github.com/ARMmbed/mbed-os/pull/10280

closed 2019-04-02 by 0xc0170