ARMmbed/mbed-os #10236: About setting to interval of BLE beacon

heejungPark, 2019-03-27:
### Description

Hi,

I have been testing BLE beacon and having a issue.
To change the advertising interval, I called the function named setAdvertisingInterval().
The definition of this function is the belows,
[image]
But when I set a parameter of this function under 100(for example 99), it doesn't work well.
Maybe, is minimum value of this parameter 100?
Is there a way to set this value under 100? (Actually, I want to set to 2ms.)

Thanks.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

vincha04, 2019-03-27:
FYI: I believe the spec-defined minimum for BLE advertising is 20ms.

heejungPark, 2019-03-28:
Hi @vincha04 

I have seen that somewhere.
But my code is not working...
I'm sorry but could you please share a code that is working well when the interval is 20ms.

Thanks.

desmond-blue, 2019-03-28:
Hi @heejungPark , the API `setAdvertisingInterval()` has [been deprecated](https://os.mbed.com/docs/mbed-os/v5.12/mbed-os-api-doxy/class_gap.html#aae6623b7ad7087f86113758e20068d9e), instead, you should config the interval with `ble::AdvertisingParamters`, please refer [here](https://os.mbed.com/docs/mbed-os/v5.12/mbed-os-api-doxy/classble_1_1_advertising_parameters.html).

And please notice that  [Questions](https://os.mbed.com/questions/) area of Mbed would be a better place for asking usage of API since this is also a good information for the whole community.

Thanks for participating this project!
Desmond

closed 2019-04-18 by ciarmcom

linlingao, 2019-04-30:
@heejungPark It appears Desmond has answered your question. Please let us know if you have further questions.

heejungPark, 2019-05-02:
@desmond-blue and @linlingao 
Thanks for your answer.
I don't have further question yet.

Thanks.